embassy-rs/trouble #32: Example for esp32

lulf, 2024-05-09:


artisdom, 2024-08-06:
Interested to get this running on esp32, not sure where to start. :)

lulf, 2024-08-06:
I suggest to look at the [bleps example](https://github.com/bjoernQ/bleps/tree/main/esp32s3-serial-hci) as the starting point. You basically you need to provide Trouble with an implementation of a SerialTransport as done [here](https://github.com/embassy-rs/trouble/blob/main/examples/serial-hci/src/main.rs#L52-L55).

The ESP32 HAL already implements the Read and Write traits required I think, but it needs to be able to 'split' them to use with SerialTransport.

Sycrosity, 2024-08-18:
@artisdom did you have any success with this?

artisdom, 2024-08-18:
@Sycrosity have been distracted by other things from this, but would be keen to try it out if anyone got it working. :)

lulf, 2024-08-20:
I've added an implementation of bt-hci traits to esp-hal https://github.com/esp-rs/esp-hal/pull/1971 , and a PR with trouble example in #81.

closed 2024-08-20 by lulf

Sycrosity, 2024-08-21:
@lulf amazing! I was attempting to work on this myself hack-ily but this is a much better result - thanks for the work!

PR #81 (2024-08-20): Add example for esp32
Fixes #32