embassy-rs/trouble #72: Example how to handle events

Syphixs, 2024-07-25:
Hey! Thank you for your great work. I am coming from bleps and wanted to understand how I can handle events in trouble. For example how do i call a function when a write on a specific characteristic happens? I am closely looking at the ble_bas_peripheral.rs example in examples/nrf-sdc but I don't understand when the event is fired as this 
```rust
    info!("Starting advertising and GATT service");
    let _ = join3(
        ble.run(),
        async {
            loop {
                info!("Reading next");
                match server.next().await {
                    Ok(event) => {
                        info!("Gatt event: {:?}", event);
                    }
                    Err(e) => {
                        error!("Error processing GATT events: {:?}", e);
                    }
                }
            }
        }, .....
```
only fires a single time at the beginning and never again. I thought i can just define a new custom service and characteristic like this:
```rust
    let mut custom_value = [0u8; 4]; // 4-byte value
    let custom_handle = {
        let mut svc = table.add_service(Service::new(0xFFFF)); 

        svc.add_characteristic(
            0xFFFE, // Custom 16-bit UUID for characteristic
            &[CharacteristicProp::Read, CharacteristicProp::Write],
            &mut custom_value,
        )
        .build()
    };
```
and then on a write event, which does work (as i can read the updated value) the event fires.
maybe I am misunderstanding something here but the example sadly does not tell me how to use it properly yet. 
I would very much appreciate some guidance.

Syphixs, 2024-07-25:
And currently the GattEvent Enum just handles Write. Is this planned to include Read as well in some time?
```rust
#[derive(Clone)]
pub enum GattEvent<'reference, 'values> {
    Write {
        connection: &'reference Connection<'reference>,
        handle: CharacteristicHandle,
        value: &'values [u8],
    },
}
```

lulf, 2024-07-25:
@Syphixs Thanks for raising the issue. I think this is probably a bug, the write event should fire on every write as you expected. I will try to reproduce it and work on a fix.

I wasn't planning on including a read event, as it would just read from the attribute table and not require any logic in the app itself. What's the use case you have in mind for the read event? I modeled this after nrf-softdevice which really only has events for Write and 'Enable Notifications' (which is handled a bit differently in trouble, because you don't have to know if it is enabled or not).

Syphixs, 2024-07-26:
Thank you! 

Well i was thinking about a simple store which holds the last x events on the chip and waits till at least 2 devices have read an entry for example. If that happened it can safely remove the entry. Maybe this is a bad idea altogether but in bleps the store example works quite well actually. I have not yet included the second device logic for now its just a queue where on an read event i push it to the Store and on an write event with the correct message id it gets deleted, so basically just an ack. 

```rust 
use heapless::Vec;
...

const MAX_EVENTS: usize = 16;
#[derive(Copy, Clone, Debug)]
struct EventData {
    msg_uuid: [u8; 16],
    timestamp: u64,
    task_uuid: [u8; 16],
}

impl EventData {
    fn new(msg_uuid_bytes: [u8; 16], timestamp: u64, task_uuid_bytes: [u8; 16]) -> Self {
        Self {
            msg_uuid: msg_uuid_bytes,
            timestamp,
            task_uuid: task_uuid_bytes,
        }
    }
    fn packet(&self) -> [u8; 40] {
        let mut data = [0u8; 40];
        data[0..16].copy_from_slice(&self.msg_uuid);
        data[16..24].copy_from_slice(&self.timestamp.to_le_bytes());
        data[24..40].copy_from_slice(&self.task_uuid);
        data
    }
}

struct EventStorage {
    events: Vec<EventData, MAX_EVENTS>,
}

impl EventStorage {
    fn new() -> Self {
        Self { events: Vec::new() }
    }

    fn add_event(&mut self, event: EventData) {
        if self.events.len() == MAX_EVENTS {
            self.events.remove(0);
        }
        self.events.push(event).unwrap();
    }

    fn remove_event(&mut self, msg_uuid: &[u8; 16]) -> Option<EventData> {
        for i in 0..self.events.len() {
            if self.events[i].msg_uuid == *msg_uuid {
                return Some(self.events.remove(i));
            }
        }
        None
    }

    fn get_oldest_event(&self) -> Option<&EventData> {
        self.events.first()
    }

    fn clear_events(&mut self) {
        self.events.clear();
    }
}
```
If you have a simpler more elegant solution I am all ears :D

Syphixs, 2024-07-26:
And here is my simple example code for easier reproducibility. Do mind that this is just for testing so I did not use everything and commented out a lot. Oh and I am using the nrf52833DK for testing.

main.rs
```rust
#![no_std]
#![no_main]
#![feature(impl_trait_in_assoc_type)]

use core::cell::RefCell;

use defmt::{error, info, unwrap};
use embassy_executor::Spawner;
use embassy_futures::join::join3;
use embassy_nrf::gpio::{Input, Level, Output, OutputDrive, Pull};
use embassy_nrf::{bind_interrupts, pac};
use embassy_sync::blocking_mutex::raw::NoopRawMutex;
use embassy_time::{Duration, Timer};
use heapless::Vec;
use nrf_sdc::mpsl::MultiprotocolServiceLayer;
use nrf_sdc::{self as sdc, mpsl};
use sdc::rng_pool::RngPool;
use static_cell::StaticCell;
use trouble_host::advertise::{
    AdStructure, Advertisement, BR_EDR_NOT_SUPPORTED, LE_GENERAL_DISCOVERABLE,
};
use trouble_host::attribute::{
    AttributeTable, CharacteristicHandle, CharacteristicProp, Service, Uuid,
};
use trouble_host::gatt::GattEvent;
use trouble_host::{Address, BleHost, BleHostResources, PacketQos};
use {defmt_rtt as _, panic_probe as _};

bind_interrupts!(struct Irqs {
    RNG => nrf_sdc::rng_pool::InterruptHandler;
    SWI0_EGU0 => nrf_sdc::mpsl::LowPrioInterruptHandler;
    POWER_CLOCK => nrf_sdc::mpsl::ClockInterruptHandler;
    RADIO => nrf_sdc::mpsl::HighPrioInterruptHandler;
    TIMER0 => nrf_sdc::mpsl::HighPrioInterruptHandler;
    RTC0 => nrf_sdc::mpsl::HighPrioInterruptHandler;
});

const DEVICE_NAME_CHAR: u16 = 0x2a00;
const APPEARANCE_CHAR: u16 = 0x2a01;
const GENERAL_SERVICE: u16 = 0x1800;
const BATTERY_SERVICE: u16 = 0x180F;
const BATTERY_LEVEL_CHAR: u16 = 0x2a19;
const EVENT_SERVICE: u16 = 0xFFFF;
const EVENT_CHAR: u16 = 0xFFFE;

const MAX_EVENTS: usize = 16;
#[derive(Copy, Clone, Debug)]
struct EventData {
    msg_uuid: [u8; 16],
    timestamp: u64,
    task_uuid: [u8; 16],
}

impl EventData {
    fn new(msg_uuid_bytes: [u8; 16], timestamp: u64, task_uuid_bytes: [u8; 16]) -> Self {
        Self {
            msg_uuid: msg_uuid_bytes,
            timestamp,
            task_uuid: task_uuid_bytes,
        }
    }
    fn packet(&self) -> [u8; 40] {
        let mut data = [0u8; 40];
        data[0..16].copy_from_slice(&self.msg_uuid);
        data[16..24].copy_from_slice(&self.timestamp.to_le_bytes());
        data[24..40].copy_from_slice(&self.task_uuid);
[... 376 lines cut ...]
trouble-host = { git = "https://github.com/embassy-rs/trouble.git", branch = "main" }

[profile.release]
debug = 2
```
.cargo/config.toml
```toml
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
# replace nRF82840_xxAA with your chip as listed in `probe-rs chip list`
runner = "probe-rs run --chip nRF52833_xxAA"

# rustflags = ["-C", "link-arg=-Tlink.x", "-C", "link-arg=-Tdefmt.x"]

[build]
target = "thumbv7em-none-eabi"
# Pick ONE of these compilation targets
# target = "thumbv6m-none-eabi"    # Cortex-M0 and Cortex-M0+
# target = "thumbv7m-none-eabi"    # Cortex-M3
# target = "thumbv7em-none-eabi"   # Cortex-M4 and Cortex-M7 (no FPU)
# target = "thumbv7em-none-eabihf" # Cortex-M4F and Cortex-M7F (with FPU)

[env]
DEFMT_LOG = "trace"
```
memory.x and build.rs are the example ones ofc with the correct settings for nrf52833

lulf, 2024-07-29:
@Syphixs Thanks for the example! I'll create an integration test for gatt so that we can be sure it will continue to work. 

Regarding the use case for the Read event, that seems valid, I just haven't seen GATT used in this way before. It should be fairly easy to add!

lulf, 2024-07-31:
@Syphixs I've added an integration test for GATT. Turns out that events were never returned :see_no_evil: I've added read events as well now, and a minor mod to write events not returning the value. You can read that from the AttributeTable you pass to the gatt server (see ble_bas_peripheral for example). 

Good news is that to write the integration test I also implemented a basic GATT client, so that's now supported as well along with an example (ble_bas_central).

That should solve the original issue, so I'm closing that. Feel free to reopen if it's not working for you.

closed 2024-07-31 by lulf