embassy-rs/trouble #90: How to change GattServers MTU Size easily?

Syphixs, 2024-08-23:
As far as I understand I can work with the underlying l2cap (protocol) to change the MTU size to my liking but its unclear to me how i set up an GATT Server with a bigger MTU Size. It seems on the GattServer struct the MTU size is a fixed value of 27 including the 4 byte header. I am mostly working with an Iphone (and an nrf52833) and I have read that the MTU size has to be negotiated with all apple products so maybe this is the real problem.

Changing the L2CAP_MTU const does indeed change the byte size which is allowed on the server itself but is not advertising this MTU size. 

Furthermore in the nrf ble_bas_peripheral example the sdc buffer  does also not get configured as far as I can see.
for clarity
```rust

fn build_sdc<'d, const N: usize>(
    p: nrf_sdc::Peripherals<'d>,
    rng: &'d RngPool,
    mpsl: &'d MultiprotocolServiceLayer,
    mem: &'d mut sdc::Mem<N>,
) -> Result<nrf_sdc::SoftdeviceController<'d>, nrf_sdc::Error> {
    sdc::Builder::new()?
        .support_adv()?
        .support_peripheral()?
        .peripheral_count(1)?
        .buffer_cfg(L2CAP_MTU as u8, L2CAP_MTU as u8, L2CAP_TXQ, L2CAP_RXQ)? // this is missing
        .build(p, rng, mpsl, mem)
}
```

lulf, 2024-08-23:
I admit the APIs for GATT MTU size is not great, so if you have any suggestions on how to improve that, that would be welcome (We don't actually use GATT at work, so it's not received a lot of attention).  

For GATT clients, the att mtu is negotiated when creating the client (see ble_bas_central), but for the gatt server, it doesn't change the 'default' MTU size used in the mtu negotiation. I think the fix would be to make the fn gatt_server()  on the BleHost type to configure this in the ConnectionManager, so that it would use that MTU when negotiation rather than always 23 (which is hardcoded if you look for that in connection_manager.rs).

Regarding the SDC configuration, it has a default l2cap mtu of 27 and 1 item for both rx and tx, which is used if you don't set anything.

Syphixs, 2024-08-23:
No problem I just want to point out all the things I find difficult to work with in this crate :D .

I like the approach but I am not sure if that is possible that easily the same way as in the gatt_client. At the time the gatt client is created a connection already exists and then the MTU exchange happens.
as seen in:
```rust
...
    info!("Scanning for peripheral...");
    let _ = join(ble.run(), async {
        let conn = unwrap!(ble.connect(&config).await);
        info!("Connected, creating gatt client");

        let mut client = ble.gatt_client::<10, 128>(&conn).await.unwrap();

// on the gatt client 
    /// Creates a GATT client capable of processing the GATT protocol using the provided table of attributes.
    #[cfg(feature = "gatt")]
    pub async fn gatt_client<'reference, const MAX: usize, const ATT_MTU: usize>(
        &'reference self,
        connection: &Connection<'reference>,
    ) -> Result<GattClient<'reference, 'd, T, MAX, ATT_MTU>, BleHostError<T::Error>> {
        let l2cap = L2capHeader { channel: 4, length: 3 };
        let mut buf = [0; 7];
        let mut w = WriteCursor::new(&mut buf);
        w.write_hci(&l2cap)?;
        w.write(att::AttReq::ExchangeMtu { mtu: ATT_MTU as u16 })?;

        let mut grant = self.acl(connection.handle(), 1).await?;

        grant.send(w.finish()).await?;

        Ok(GattClient {
            services: heapless::Vec::new(),
            rx: self.att_inbound.receiver().into(),
            ble: self,
            connection: connection.clone(),
        })
    }

```

As we need the gatt server already initialized before we start out main loop(s) and have no connections yet we need to configure it on the ConnectionManager side as you said, but its unclear to me when the actual MTU Exchange then should take place?

PR #91 (2024-08-24): support configuring gatt server mtu
Resolves #90
closed 2024-08-24 by lulf