embassy-rs/trouble #95: How to add descriptor to characteristic?

HaoboGu, 2024-08-26:
In current implementation, `add_descriptor` returns a `DescriptorHandle`, which doesn't have `build()` like `CharacteristicBuilder`. That means I cannot do things like:

```rust
let input_keyboard_handle = table
    .add_service(Service::new(BleSpecification::HidService as u16))
    .add_characteristic(
        BleCharacteristics::HidReport as u16,
        &[
            CharacteristicProp::Read,
            CharacteristicProp::Write,
            CharacteristicProp::Notify,
        ],
        &mut input_keyboard_data,
    )
    .add_descriptor(
        BleDescriptor::ReportReference as u16,
        &[
            CharacteristicProp::Read,
            CharacteristicProp::Write,
            CharacteristicProp::Notify,
        ],
        &mut input_keyboard_desc_data,
    )
    .build();
```

I can only get a `DescriptorHandle` without using `build`, which cannot be used to be compared using

```rust
  match server.next().await {
      Ok(GattEvent::Read {
          handle,
          connection: _,
      }) => {
          if handle == input_keyboard_handle {       <= ERROR: mismatched types: expected `Characteristic`, found `DescriptorHandle`
              info!("Input keyboard read");
          }
          ...
   }
```

lulf, 2024-08-26:
Thanks for raising the issue. The descriptor api is abit weird as it is, I'm not sure if it's really needed. I think you can work around it like this:

```
let mut svc = table
    .add_service(Service::new(BleSpecification::HidService as u16));

let input_kbd_data_h = svc.add_characteristic(
        BleCharacteristics::HidReport as u16,
        &[
            CharacteristicProp::Read,
            CharacteristicProp::Write,
            CharacteristicProp::Notify,
        ],
        &mut input_keyboard_data,
    ).build();

let input_kbd_desc_h = svc.add_descriptor(
        BleDescriptor::ReportReference as u16,
        &[
            CharacteristicProp::Read,
            CharacteristicProp::Write,
            CharacteristicProp::Notify,
        ],
        &mut input_keyboard_desc_data,
    )
    .build();
```

lulf, 2024-08-26:
I removed the api to add descriptor, you should just use the add_characteristic and add_characteristic_ro to add those to the ServiceBuilder instead.

HaoboGu, 2024-08-27:
thanks! I'll try it soon:D

HaoboGu, 2024-08-27:
I tried the latest commit, the following is my initialization code:

```rust
let mut hid_svc = table.add_service(Service::new(BleSpecification::HidService as u16));
let mut input_keyboard_data = [0u8; 8];
let input_keyboard_handle = hid_svc.add_characteristic(
    BleCharacteristics::HidReport as u16,
    &[
        CharacteristicProp::Read,
        CharacteristicProp::Write,
        CharacteristicProp::Notify,
    ],
    &mut input_keyboard_data,
);
drop(hid_svc);
let server = ble.gatt_server::<NoopRawMutex, 50, L2CAP_MTU>(&table);
```

But I got the following error:

```
error[E0597]: `input_keyboard_data` does not live long enough
   --> /Users/haobogu/Projects/keyboard/rmk/rmk/src/ble/trouble.rs:106:9
    |
98  |     let mut input_keyboard_data = [0u8; 8];
    |         ----------------------- binding `input_keyboard_data` declared here
...
106 |         &mut input_keyboard_data,
    |         ^^^^^^^^^^^^^^^^^^^^^^^^ borrowed value does not live long enough
...
204 | }
    | -
    | |
    | `input_keyboard_data` dropped here while still borrowed
    | borrow might be used here, when `hid_svc` is dropped and runs the `Drop` code for type `trouble_host::attribute::ServiceBuilder`
    |
    = note: values in a scope are dropped in the opposite order they are defined
```

What's the recommended way to fill the data buf here? I tried `StaticCell`, it looks kind of weird though. 

```rust
// What I tried
static input_keyboard_data: StaticCell<[u8; 8]> = StaticCell::new();
let input_keyboard_handle = hid_svc.add_characteristic(
    BleCharacteristics::HidReport as u16,
    &[
        CharacteristicProp::Read,
        CharacteristicProp::Write,
        CharacteristicProp::Notify,
    ],
    input_keyboard_data.init([0u8; 8]),
);
```

HaoboGu, 2024-08-27:
Yes, I used the same example, what I did is just adding hid svc. 

I found how to repro the error: storage data MUST be declared BEFORE the `.add_service`, otherwise the error occurs:

```
    // Error!: `bat_level` does not live long enough
    let mut bat_svc = table.add_service(Service::new(0x180f));
    let mut bat_level = [23; 1];
    let level_handle = bat_svc.add_characteristic(
        0x2a19,
        &[CharacteristicProp::Read, CharacteristicProp::Notify],
        &mut bat_level,
    );
    bat_svc.build();

    // Compiles
    let mut bat_level = [23; 1];
    let mut bat_svc = table.add_service(Service::new(0x180f));
    let level_handle = bat_svc.add_characteristic(
        0x2a19,
        &[CharacteristicProp::Read, CharacteristicProp::Notify],
        &mut bat_level,
    );
    bat_svc.build();
```

I have no idea about why the order matters in this case.

> Fwiw, the drop() thing is a bit weird, but the builder must be consumed to guarantee no &mut refs to the table. If you look at the example, the `.build()' on the svc does that too.

Thanks for pointing it out! I've changed all to `.build()`.

lulf, 2024-08-27:
Ah, yes. I think the compiler is just assigning them different lifetimes rather than 'the same' for some reason. Could be that trouble trips it up by introducing too many lifetimes in the gatt case :sweat_smile:

HaoboGu, 2024-08-27:
That's reasonable. I'm closing this issue since the problem has been solved. Thanks again for your kind reply!

closed 2024-08-27 by HaoboGu

Univa, 2024-09-02:
Hey folks! Noticed the descriptor API was removed, which brought me here.

> I'm not sure if it's really needed. I think you can work around it like this:

@lulf do you mind explaining how this workaround works? I tried something similar to the workaround example you gave, and I couldn't get descriptors to show up:

(nRF connect app)
[image]

Previously, the `add_descriptor` function would avoid pushing the [`Attribute` with `uuid: CHARACTERISTIC_UUID16`](https://github.com/embassy-rs/trouble/blob/e6d2b295ef67e3b489db3d0e0e1bcfff05f77f57/host/src/attribute.rs#L427), which allows us to get descriptors. Code like this would work:

```rust
let mut hid_service = table.add_service(Service::new(0x1812));
let mut characteristic = hid_service.add_characteristic(
    0x2a4d,
    &[
        CharacteristicProp::Read,
        CharacteristicProp::Write,
        CharacteristicProp::Notify,
    ],
    &mut data,
);
let descriptor_handle = characteristic.add_descriptor_ro(0x2908, &[0x01, 0x01]);
let characteristic_handle = characteristic.build();
drop(hid_service); // drop &mut ref to `table`

let server = ble.gatt_server(&table); 

// you can use `descriptor_handle` and `report_handle` as you want
```

[image]

lulf, 2024-09-02:
Oooh... so I think I understand what you were using it for now, which I didn't at the time. So, basically it allows adding _multiple_ descriptors for a characteristic. That is a use case I overlooked, sorry about that! https://github.com/embassy-rs/trouble/pull/104