embassy-rs/trouble #101: Support for nrf5340 dual core architecture

igiona, 2024-08-31:
I'm quite familiar with the nrf52 family and I used the SoftDevice extensively in C and I did some fairly simple work in Rust using embassy-softdevice.
Recently I started looking into the nrf5340, and I though giving `trouble` a shot 'cause Nordic doesn't support the nrf53 family within the SoftDevice stack anymore :'(

I never worked on a dual core microcontroller in Rust, hence I'm not sure where to start here?
I saw that there  is some support for the nrf5340 chip within embassy, but the examples do not really showcase the usage of the double cores.

Any suggestion on how to get started with trouble on that chip?

lulf, 2024-09-02:
I think with the nrf53, you can run the controller on the network core, and then trouble on the application core, communicating over the IPC bus. You'd need to implement the traits from `bt-hci` that send/receive HCI packets to the network core.

igiona, 2024-09-02:
> I think with the nrf53, you can run the controller on the network core, and then trouble on the application core, communicating over the IPC bus. You'd need to implement the traits from `bt-hci` that send/receive HCI packets to the network core.

Are you aware of any non-sdc controller available for the nrf5340?

Why should trouble run on the app-core in your opinion?
I would have rather placed it in the net-core and I would have communicated over IPC only at an higher level of abstraction (like GATT read/write/notifications).
The app-core is quite power hungry, hence I wouldn't bothered waking it app at every "low-level" BLE event.

In general, I'm still not quite sure how to handle 2 applications in one rust project... would you handled them as compleatly separated apps?

lulf, 2024-09-02:
> Are you aware of any non-sdc controller available for the nrf5340?

I'm not very familiar with it, but my impression was that Nordic provides a controller firmware that you flash to the network core and that's all you need.

> Why should trouble run on the app-core in your opinion? I would have rather placed it in the net-core and I would have communicated over IPC only at an higher level of abstraction (like GATT [...]

Keep in mind even Trouble does not get notified of low level events. If you're using only GATT, there will not be any 'hidden' events on a connection other than the initial 'connected', 'gatt events' and 'disconnected', so I doubt you would save a lot of power by running trouble on the net core (and more likely complicate your firmware build/deployment model).

> In general, I'm still not quite sure how to handle 2 applications in one rust project... would you handled them as completely separated apps that communicate over an IPC API?

I think you'd generally treat them as separate build targets/firmware, but they may share some common crates for type definitions over the IPC etc.

igiona, 2024-09-02:
Thanks for the inputs!

> I'm not very familiar with it, but my impression was that Nordic provides a controller firmware that you flash to the network core and that's all you need.

Your impression was right: https://github.com/embassy-rs/trouble/issues/101#issuecomment-2325045028
[image]

To me it still seems like a waste of resources, to run only the controller on a dedicated CPU, in the nrf52 series app and BLE are coexisting with pretty decent performances after all... but I might underestimate all the job that the controller (and multi-protocol stuff) need to do...

I'll keep you posted here with any progress on this...

igiona, 2025-02-07:
That's my understanding too.

Trouble is, you need first to implement Open amp RPMSG protocol in order to be able to talk to the net-core running the BLE controller.

I started gathering some documents etc. But I must say that the documentation I found so far isn't that great.

canatella, 2025-02-13:
Yep, been looking a bit around too and indeed, documentation is a bit lacking... Most interesting things I found is https://github.com/nxp-mcuxpresso/rpmsg-lite/tree/main/lib. It looks small enough as a basis to understand this ipc stuff is working. I'll see if I can get to that at some point.

0e4ef622, 2025-02-26:
Zephyr supports other IPC backends like [ICMsg](https://docs.zephyrproject.org/latest/services/ipc/ipc_service/backends/ipc_service_icmsg.html) and [ICBMsg](https://docs.zephyrproject.org/latest/services/ipc/ipc_service/backends/ipc_service_icbmsg.html), might be simpler to implement. Also somewhat relevant https://github.com/apache/mynewt-nimble/pull/1882