embassy-rs/trouble #107: Potential issue in GATT processing using Windows LE explorer

lulf, 2024-09-05:
See https://github.com/esp-rs/esp-hal/pull/1971#issuecomment-2311760670 for a packet capture. Attempt to reproduce with other HCI controllers as well to see if it's esp-specific.

timokroeger, 2025-02-26:
I see the same problem trouble running on a nrf52 device.
Any tool using the WinRT bluetooth API gets stuck when trying to discover services (that includes the Web Bluetooth API in Edge browser).

Here are the defmt logs of my device:
```
[host] filter accept list size: 8
[host] setting txq to 1, fragmenting at 255
[host] configuring host buffers (8 packets of size 255)
[host] initialized
[host] enabling advertising
[ble] advertising, interval: Duration { ticks: 656 }
[host] connection with handle ConnHandle(0) established to BdAddr([58, 72, 9a, 70, 3b, 2c])
[link][poll_accept] connection handle ConnHandle(0) in role Peripheral accepted
[host] agreed att MTU of 251
... nothing happening anymore ...
[host] disconnection event on handle 0, reason: Remote User Terminated Connection
```

Using Web Bluetooth API with chrome on android everything works fine.

timokroeger, 2025-02-26:
Windows:
[image]

Android:
[image]

commit 1b5b70b0 (2025-02-26): Remove invalid field from FindTypeValue response

This incorrect field prevented the WinRT Bluetooth API to work with
TrouBLE devices. Fixes #107.
closed 2025-02-26 by lulf