embassy-rs/trouble #108: How to manage reconnection?

Syphixs, 2024-09-09:
After a disconnection event happens (eg. phone is out of range) what would be a smart way to implement an auto-reconnect feature? Should it be as simple as restarting the advertise_task?

lulf, 2024-09-09:
Yes, but how simple you can do this depends on how many connections you want. If you want only 1 connection, you simply have a loop with advertise, accept connection, do stuff, and continue if connection is lost (advertise will start again).

If you want to handle > 1 connection, you may need to define an async task with `embassy_executor::task(pool_size = 4)`, and spawn that with the connection handle you get from accept(), and let advertise continue to run until a new connection is established.

HaoboGu, 2024-09-09:
Does someone work on the pairing manager or security manager? e.g. saving the bonding information and auto-reconnect when the device is detected.

Syphixs, 2024-09-09:
alright thank you for clarification. Currently I am only working with one connection but if the use case arises I will try your suggested executor::task approach.

closed 2024-09-10 by Syphixs