embassy-rs/trouble #109: Defmt build

okhsunrog, 2024-09-10:
The crate fails to build for me with "defmt" feature enabled. I did set default-features to false, so "log" is not enabled. Tested with commit 29dabe756dc6755a0294432e63ab1da68ec0dde0 (currently the latest one)

okhsunrog, 2024-09-10:
@lulf I'm trying to build for esp32-c3, using nightly latest toolchain. Here's my Cargo.toml:

```
[package]
name         = "esp_hal_probe"
version      = "0.1.0"
edition      = "2021"
rust-version = "1.76.0"

[dependencies]
bt-hci           = "0.1.0"
defmt            = "0.3"
defmt-rtt        = "0.4"
embassy-executor = { version = "0.6",  features = ["defmt"] }
embassy-time     = { version = "0.3",  features = ["generic-queue-8"] }
esp-alloc        = "0.4.0"
esp-backtrace    = { version = "0.14", features = ["esp32c3", "defmt", "exception-handler", "panic-handler"] }
esp-hal          = { version = "0.20", features = ["esp32c3", "defmt"] }
esp-hal-embassy  = { version = "0.3",  features = ["esp32c3", "defmt"] }
esp-wifi         = { version = "0.9",  default-features = false, features = ["defmt", "esp32c3", "ble", "async", "esp-alloc"] }
# add defmt to trouble-host when fixed
trouble-host     = { default-features = false, features = ["gatt" ], git = "https://github.com/embassy-rs/trouble", package = "trouble-host", rev = "29dabe756dc6755a0294432e63ab1da68ec0dde0"}

[profile.release]
codegen-units    = 1
debug            = 2
debug-assertions = false
incremental      = false
opt-level        = 3
lto = 'fat'
overflow-checks  = false

[patch.crates-io]
esp-hal = { git = "https://github.com/esp-rs/esp-hal", rev = "e1c27f1b22e4300ecd4d7bea970f4f72d3ba632e" }
esp-hal-embassy = { git = "https://github.com/esp-rs/esp-hal", rev = "e1c27f1b22e4300ecd4d7bea970f4f72d3ba632e" }
esp-backtrace = { git = "https://github.com/esp-rs/esp-hal", rev = "e1c27f1b22e4300ecd4d7bea970f4f72d3ba632e" }
esp-wifi = { git = "https://github.com/esp-rs/esp-hal", rev = "e1c27f1b22e4300ecd4d7bea970f4f72d3ba632e" }
esp-alloc = { git = "https://github.com/esp-rs/esp-hal", rev = "e1c27f1b22e4300ecd4d7bea970f4f72d3ba632e" }
```

It looks correct to me. Did I miss something? Here's the error I get:
```
cargo build --release
   Compiling trouble-host v0.1.0 (https://github.com/embassy-rs/trouble?rev=29dabe756dc6755a0294432e63ab1da68ec0dde0#29dabe75)
error[E0277]: the trait bound `embassy_time::Duration: Format` is not satisfied
  --> /home/okhsunrog/.cargo/git/checkouts/trouble-148cf413cadab653/29dabe7/host/src/advertise.rs:37:38
   |
37 | #[cfg_attr(feature = "defmt", derive(defmt::Format))]
   |                                      ^^^^^^^^^^^^^ the trait `Format` is not implemented for `embassy_time::Duration`, which is required by `AdvertisementParameters: Format`
   |
   = help: the following other types implement trait `Format`:
             &T
             &mut T
             ()
             (T0, T1)
             (T0, T1, T2)
             (T0, T1, T2, T3)
             (T0, T1, T2, T3, T4)
             (T0, T1, T2, T3, T4, T5)
           and 502 others
note: required for `AdvertisementParameters` to implement `Format`
  --> /home/okhsunrog/.cargo/git/checkouts/trouble-148cf413cadab653/29dabe7/host/src/advertise.rs:65:38
   |
65 | #[cfg_attr(feature = "defmt", derive(defmt::Format))]
   |                                      ^^^^^^^^^^^^^ unsatisfied trait bound introduced in this `derive` macro
66 | #[derive(Copy, Clone, Debug)]
67 | pub struct AdvertisementParameters {
   |            ^^^^^^^^^^^^^^^^^^^^^^^
   = help: see issue #48214
   = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)
help: add `#![feature(trivial_bounds)]` to the crate attributes to enable
  --> /home/okhsunrog/.cargo/git/checkouts/trouble-148cf413cadab653/29dabe7/host/src/lib.rs:11:1
   |
11 + #![feature(trivial_bounds)]
   |

error[E0277]: the trait bound `embassy_time::Duration: Format` is not satisfied
  --> /home/okhsunrog/.cargo/git/checkouts/trouble-148cf413cadab653/29dabe7/host/src/advertise.rs:65:38
   |
65 | #[cfg_attr(feature = "defmt", derive(defmt::Format))]
   |                                      ^^^^^^^^^^^^^ the trait `Format` is not implemented for `embassy_time::Duration`
   |
   = help: the following other types implement trait `Format`:
             &T
             &mut T
             ()
             (T0, T1)
             (T0, T1, T2)
             (T0, T1, T2, T3)
             (T0, T1, T2, T3, T4)
             (T0, T1, T2, T3, T4, T5)
           and 502 others
   = help: see issue #48214
   = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)
help: add `#![feature(trivial_bounds)]` to the crate attributes to enable
  --> /home/okhsunrog/.cargo/git/checkouts/trouble-148cf413cadab653/29dabe7/host/src/lib.rs:11:1
   |
11 + #![feature(trivial_bounds)]
   |

For more information about this error, try `rustc --explain E0277`.
error: could not compile `trouble-host` (lib) due to 2 previous errors
```

It builds without errors when I disable "defmt" feature for trouble crate

lulf, 2024-09-11:
Ah, it's because trouble-host is not adding defmt to embedded-time. To work around this, enable the 'defmt' feature in your embedded-time dependency.

lulf, 2024-09-11:
@okhsunrog This should be fixed now with https://github.com/embassy-rs/trouble/commit/d9d2e80b6b2ed6c3aeae1de534de557906353522 - please re-open if not.

closed 2024-09-11 by lulf

okhsunrog, 2024-09-11:
It works, thanks!