embassy-rs/trouble #115: Connection issue on iOS/macOS device

HaoboGu, 2024-09-18:
The issue is from https://github.com/embassy-rs/trouble/issues/82#issuecomment-2347925811 .

I did some more investigation, on both my own implementation and examples in latest version. I can reproduce the issue on latest commit d9d2e80b6b2ed6c3aeae1de534de557906353522:

when connecting to iOS/macOS devices, there's an error reported: `[gatt] Error processing GATT events: NotFound`

Then, the device is connected, but not functional. Then the host would terminate the connection. The following is the full trace log:

```shell
$ cargo run --release --bin ble_bas_peripheral     
   Compiling embassy-nrf v0.2.0 (https://github.com/embassy-rs/embassy.git?rev=1cfd5370ac012814b7b386ba9ad8499529bdde4e#1cfd5370)
   Compiling nrf-mpsl v0.1.0 (https://github.com/alexmoon/nrf-sdc.git?rev=3702af909d31cd81c62f15e1aa9d5f637ec935fa#3702af90)
   Compiling nrf-sdc v0.1.0 (https://github.com/alexmoon/nrf-sdc.git?rev=3702af909d31cd81c62f15e1aa9d5f637ec935fa#3702af90)
   Compiling trouble-nrf-sdc-examples v0.1.0 (/Users/haobogu/Projects/rust/trouble/examples/nrf-sdc)
    Finished `release` profile [optimized + debuginfo] target(s) in 3.63s
     Running `probe-rs run --chip nRF52840_xxAA target/thumbv7em-none-eabihf/release/ble_bas_peripheral`
      Erasing ✔ [00:00:03] [##########################################################################################################################################]      0 B/     0 B @      0 B/s (eta 0s )
  Programming ✔ [00:00:05] [#####################################################################################################################################] 152.00 KiB/152.00 KiB @ 28.90 KiB/s (eta 0s )    Finished in 5.2652664s
0.000061 WARN  Memory buffer too big. 2168 bytes needed
└─ nrf_sdc::sdc::{impl#3}::build @ /Users/haobogu/.cargo/git/checkouts/nrf-sdc-52477ea9d1e36c44/3702af9/nrf-sdc/src/fmt.rs:151 
0.000762 INFO  Our address = Address { kind: AddrKind(1), addr: BdAddr([65, 90, 227, 30, 131, 231]) }
└─ trouble_example_apps::ble_bas_peripheral::run::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
0.000915 INFO  Starting advertising and GATT service
└─ trouble_example_apps::ble_bas_peripheral::run::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
0.001007 INFO  [host] filter accept list size: 8
└─ trouble_host::host::{impl#2}::run_with_handler::{async_fn#0}::{async_block#0} @ /Users/haobogu/Projects/rust/trouble/host/src/fmt.rs:143 
0.001037 INFO  [host] setting txq to 3
└─ trouble_host::host::{impl#2}::run_with_handler::{async_fn#0}::{async_block#0} @ /Users/haobogu/Projects/rust/trouble/host/src/fmt.rs:143 
0.001098 INFO  [host] configuring host buffers (8 packets of size 251)
└─ trouble_host::host::{impl#2}::run_with_handler::{async_fn#0}::{async_block#0} @ /Users/haobogu/Projects/rust/trouble/host/src/fmt.rs:143 
0.001129 INFO  [host] initialized
└─ trouble_host::host::{impl#2}::run_with_handler::{async_fn#0}::{async_block#0} @ /Users/haobogu/Projects/rust/trouble/host/src/fmt.rs:143 
0.001220 INFO  [adv] advertising
└─ trouble_example_apps::ble_bas_peripheral::advertise_task::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
0.001312 TRACE [host] enabling advertising
└─ trouble_host::host::{impl#2}::advertise::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/host/src/fmt.rs:117 
21.151184 TRACE [host] connection with handle ConnHandle(0) established to BdAddr([15, c9, bd, 67, 78, 69])
└─ trouble_host::host::{impl#2}::handle_connection @ /Users/haobogu/Projects/rust/trouble/host/src/fmt.rs:117 
21.151306 TRACE [link][poll_accept] connection handle ConnHandle(0) in role Peripheral accepted
└─ trouble_host::connection_manager::{impl#1}::poll_accept @ /Users/haobogu/Projects/rust/trouble/host/src/fmt.rs:117 
21.151367 INFO  [adv] connection established
└─ trouble_example_apps::ble_bas_peripheral::advertise_task::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
21.272308 WARN  [host] unsupported l2cap channel id 58
└─ trouble_host::host::{impl#2}::handle_acl @ /Users/haobogu/Projects/rust/trouble/host/src/fmt.rs:156 
21.272338 WARN  [host] encountered error processing ACL data for ConnHandle(0): NotSupported
└─ trouble_host::host::{impl#2}::run_with_handler::{async_fn#0}::{async_block#2} @ /Users/haobogu/Projects/rust/trouble/host/src/fmt.rs:156 
21.302276 TRACE [host] agreed att MTU of 185
└─ trouble_host::host::{impl#2}::handle_acl @ /Users/haobogu/Projects/rust/trouble/host/src/fmt.rs:117 
21.392547 INFO  [gatt] Read event on Characteristic { cccd_handle: Some(35), handle: 34 }
└─ trouble_example_apps::ble_bas_peripheral::gatt_task::{async_fn#0}::{closure#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
21.452545 ERROR [gatt] Error processing GATT events: NotFound
└─ trouble_example_apps::ble_bas_peripheral::gatt_task::{async_fn#0}::{closure#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:169 
23.151428 INFO  [adv] notifying connection of tick 1
└─ trouble_example_apps::ble_bas_peripheral::advertise_task::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
25.151519 INFO  [adv] notifying connection of tick 2
└─ trouble_example_apps::ble_bas_peripheral::advertise_task::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
27.151611 INFO  [adv] notifying connection of tick 3
└─ trouble_example_apps::ble_bas_peripheral::advertise_task::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
29.151702 INFO  [adv] notifying connection of tick 4
└─ trouble_example_apps::ble_bas_peripheral::advertise_task::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
31.151794 INFO  [adv] notifying connection of tick 5
└─ trouble_example_apps::ble_bas_peripheral::advertise_task::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
33.151885 INFO  [adv] notifying connection of tick 6
└─ trouble_example_apps::ble_bas_peripheral::advertise_task::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
35.151977 INFO  [adv] notifying connection of tick 7
└─ trouble_example_apps::ble_bas_peripheral::advertise_task::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
37.152069 INFO  [adv] notifying connection of tick 8
└─ trouble_example_apps::ble_bas_peripheral::advertise_task::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
39.152160 INFO  [adv] notifying connection of tick 9
└─ trouble_example_apps::ble_bas_peripheral::advertise_task::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
41.152252 INFO  [adv] notifying connection of tick 10
└─ trouble_example_apps::ble_bas_peripheral::advertise_task::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
43.152343 INFO  [adv] notifying connection of tick 11
└─ trouble_example_apps::ble_bas_peripheral::advertise_task::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
45.152435 INFO  [adv] notifying connection of tick 12
└─ trouble_example_apps::ble_bas_peripheral::advertise_task::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
47.152526 INFO  [adv] notifying connection of tick 13
└─ trouble_example_apps::ble_bas_peripheral::advertise_task::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
49.152618 INFO  [adv] notifying connection of tick 14
└─ trouble_example_apps::ble_bas_peripheral::advertise_task::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
51.152709 INFO  [adv] notifying connection of tick 15
└─ trouble_example_apps::ble_bas_peripheral::advertise_task::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
51.483886 INFO  [host] disconnection event on handle 0, reason: Remote User Terminated Connection
└─ trouble_host::host::{impl#2}::run_with_handler::{async_fn#0}::{async_block#2} @ /Users/haobogu/Projects/rust/trouble/host/src/fmt.rs:143 
53.152801 INFO  [adv] notifying connection of tick 16
└─ trouble_example_apps::ble_bas_peripheral::advertise_task::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
53.152862 INFO  [adv] advertising
└─ trouble_example_apps::ble_bas_peripheral::advertise_task::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/examples/apps/src/fmt.rs:143 
53.152923 TRACE [host] disabling advertising
└─ trouble_host::host::{impl#2}::run_with_handler::{async_fn#0}::{async_block#0} @ /Users/haobogu/Projects/rust/trouble/host/src/fmt.rs:117 
53.153045 TRACE [host] enabling advertising
└─ trouble_host::host::{impl#2}::advertise::{async_fn#0} @ /Users/haobogu/Projects/rust/trouble/host/src/fmt.rs:117 
```

lulf, 2024-09-18:
May I ask, which client are you using to connect? I'm not able to reproduce with the nrf connect app on iOS

HaoboGu, 2024-09-18:
> May I ask, which client are you using to connect? I'm not able to reproduce with the nrf connect app on iOS

I'm using nrf connect as well. Here is the client log:

[image]

HaoboGu, 2024-09-18:
After the last callback, the device is disconnected.

lulf, 2024-09-18:
Hmm, maybe the discovery process is different in some cases. Which app version are you using?

HaoboGu, 2024-09-18:
My nrf connect's application version is 2.7.11, build number is 26

lulf, 2024-09-18:
Looks like my old ios didn't get newer nrf connect. But luckily I was able to reproduce your issue on a mac.

lulf, 2024-09-18:
@HaoboGu The fix merged in #116 fixes the issue for me. Please reopen if you're still seeing the issue with that.

closed 2024-09-18 by lulf

martintv, 2024-09-18:
@HaoboGu : about the 'SoftdeviceController: 20:201' crash:

is it easy for you to reproduce it? you are connected to a mac as peripheral when it happens? any way you could get a sniffer-log of on air trafiic when it happens?

HaoboGu, 2024-09-18:
> @HaoboGu : about the 'SoftdeviceController: 20:201' crash:
> 
> is it easy for you to reproduce it? you are connected to a mac as peripheral when it happens? any way you could get a sniffer-log of on air trafiic when it happens?

Yes, I can repro it everytime after I click "connect" to the peripheral. I'll try to get a sniffer-log.

igiona, 2024-09-19:
> I'm not sure what's required for a BLE device to be listed in the Mac OS list. Could be there are some special characteristics that you need to provide for it to get listed. From what I gather, it might only show devices that implement the officially supported bluetooth profiles

That's correct, apple is quite picky& demanding (the same is also valid for iOS).
If you scan your deceives in an app you'll get the device listed regardless what service you provide, but not from the OS menu. There you need to provide all they want you to provide.

HaoboGu, 2024-09-19:
@martintv I tried to sniffer the BLE traffic using [nRF sniffer](https://docs.nordicsemi.com/bundle/nrfutil_ble_sniffer_pdf/resource/nRF_Sniffer_BLE_UG_v4.0.0.pdf). Here is the wireshark log file, it can be opened using wireshark after installing nRF sniffer:

[nrf_sniffer.pcapng.zip](https://github.com/user-attachments/files/17059887/nrf_sniffer.pcapng.zip)

I hope it's useful :)

lulf, 2024-09-20:
@HaoboGu Hey, the feedback I got from Martin is that the data that trouble attempts to send is too big. What is the BleHostResources configuration you are using, and what is the size of the GATT attributes? It's likely related to either misconfigured limits, or trouble not segmenting the gatt payloads appropriately.

HaoboGu, 2024-09-20:
@lulf My full configuration is at https://github.com/HaoboGu/rmk/blob/feat/trouble/rmk/src/ble/trouble.rs, I set `MAX_ATTRIBUTES` to 50, which is a relatively large number. But the number of added attribute is actually less than I used in `nrf-softdevice`(see nrf-softdevice [example](https://github.com/embassy-rs/nrf-softdevice/blob/master/examples/src/bin/ble_keyboard_peripheral_builder.rs))

PR #116 (2024-09-18): Return characteristic even if next was not cccd
Fixes #115
commit 3135b3b8 (2024-09-18): Return characteristic even if next was not cccd

Fixes #115