embassy-rs/trouble #135: Requirements for implementing the Matter BTP protocol on top of `trouble-host`

ivmarkov, 2024-10-09:
(**IN PROGRESS**)

I'm currently researching for a way to layer the **Matter BTP protocol** (section "4.18. Bluetooth Transport Protocol (BTP)" of the Matter Core Spec V1.3) on top of `trouble-host`.

(
- You can download the spec upon request from the CSA website [here](https://csa-iot.org/developer-resource/specifications-download-request/
- I'll refer below to the `rs-matter` codebase too, as it is available [online](), and has an [out-of-the-box implementation on top of BlueZ](https://github.com/project-chip/rs-matter/blob/main/rs-matter/src/transport/network/btp/gatt/bluer.rs), as well as an ESP-IDF Bluedroid [implementation](https://github.com/ivmarkov/esp-idf-matter/blob/ba27ed0e9763481b64f06ca8a8f588e623ef2560/src/ble.rs#L170)
).

A very short, informal summary of what are the BTP protocol requirements w.r.t. a GATT Server: it boils down to modeling a duplex read/write pipe on top of two characteristics (referred to as `C1` and `C2`) in the BTP spec:
- [C1](https://github.com/project-chip/rs-matter/blob/3bf4f7980103700e7b8f51d77281d5c661761bbc/rs-matter/src/transport/network/btp/gatt.rs#L39) has a [`Write`-Confirmed permission](https://github.com/project-chip/rs-matter/blob/3bf4f7980103700e7b8f51d77281d5c661761bbc/rs-matter/src/transport/network/btp/gatt/bluer.rs#L178), and serves as the "read" end of the pipe, from the POV of the device which is being commissioned (the GATT server), and as the "write" end of the pipe from the point of view of the Commissioner (your phone)
- [C2](https://github.com/project-chip/rs-matter/blob/3bf4f7980103700e7b8f51d77281d5c661761bbc/rs-matter/src/transport/network/btp/gatt.rs#L41) is a [CCCD "indicate" characteristics](https://github.com/project-chip/rs-matter/blob/main/rs-matter/src/transport/network/btp/gatt/bluer.rs#L205) that serves as the "write" end of the pipe, from the POV of the device which is being commissioned

In terms of requirements around these two characteristics:
- W.r.t. `C1` we need the Matter BTP implementation code to be notified about _every single incoming write request_, with the following data:
  - Peer BT address (possibly coming from the GATT connection itself)
  - Data offset
  - Data length
  - Data octets
  - Negotiated MTU
  - Moreover, the notifications should arrive in the user code (Matter BTP implementation) _sequentially_ one after the other, so that the BTP implementation has the chance to inspect each incoming data chunk separately
- W.r.t. `C2`:
  - We need to be able to notify (or rather "indicate") a remote peer - at any point in time - with a slice of data octets
  - We also need to know when a remote peer is subscribing/unsubscribing to/from characteristic `C2`

Additionally, we must be able to advertise [our own vendor-specific payload](https://github.com/project-chip/rs-matter/blob/3bf4f7980103700e7b8f51d77281d5c661761bbc/rs-matter/src/transport/network/btp/gatt.rs#L56), which is described in "5.4.2.5.6. Advertising Data" of the Matter Core Spec.

In terms of how a GATT "peripheral", specific to the purposes of the BTP protocol might look like, you can look [here](https://github.com/project-chip/rs-matter/blob/3bf4f7980103700e7b8f51d77281d5c661761bbc/rs-matter/src/transport/network/btp/gatt.rs#L178). (Do note that whether the `C` writes are coming via a blocking callback or are somehow "pulled" using `async` next external style iteration is not set in stone and we can change that.)

lulf, 2024-10-14:
* Regarding confirmed writes, that is not supported by trouble-host at this point, but should probably not be a lot of work since I think it's a matter of using a different form of response to the att request.
* Indication is basically notification without ACK, and should be implementable in a similar manner as notifications today (but probably with different CCCD configuration as well as different payload).

alexmoon, 2024-10-14:
>     * Indication is basically notification without ACK, and should be implementable in a similar manner as notifications today (but probably with different CCCD configuration as well as different payload).

Just a real minor nit, but for clarity, indication's have ACKs and notifications do not. It is also possible to have both notifications and indications allowed for a characteristic and the client could enable one or the other (or both).

petekubiak, 2025-01-13:
I think it's probably worth revisiting whether there are still outstanding requirements from this issue, now that #200 is merged.

@ivmarkov would you mind taking another look at this in light of the recent changes which have been made to trouBLE and let us know if there are further features which are needed to fully address it?

ivmarkov, 2025-01-13:
> I think it's probably worth revisiting whether there are still outstanding requirements from this issue, now that #200 is merged.
>
> @ivmarkov would you mind taking another look at this in light of [...]

I need a couple of weeks or so to get back to this, as I was completely underwater during the last three months.

Is TrouBLE also supporting indications in the meantime? And confirmed writes?

petekubiak, 2025-01-23:
Sorry I didn't reply to this. I think support for these two features is still in the pipeline but I think you mentioned elsewhere that you've managed to hack your way around that for now?

ivmarkov, 2025-01-23:
> Sorry I didn't reply to this. I think support for these two features is still in the pipeline but I think you mentioned elsewhere that you've managed to hack your way around that for now?

I managed to hack my way around by just sending notifications-as-indications and not sending ACKs. This somehow works on the ESP, but I'm testing with the cyw43 right now, and it seems to fail miserably there. Haven't gotten to the root cause there yet, but it _seems_ to me that the driver is re-sending the same packages. Probably because I don't ACK them.

I think until we have proper confirmed writes and indications, this issue has to stay open. And we need a successfull `rs-matter` run on anything else except ESPs. Like, RP and the cyw43.

petekubiak, 2025-01-23:
Agreed, I definitely don't think this issue should be closed until indications and confirmed writes are implemented

ivmarkov, 2025-02-05:
> Agreed, I definitely don't think this issue should be closed until indications and confirmed writes are implemented

The status on this issue is as follows:
- Confirmed writes were actually implemented from the get go (blush); I only got to know that once I digged a bit deeper in the ATT protocol due to #267 
- Indications (and their confirmations) are still _not_ implemented in the GATT server layer, but I - in fact - I do not need these to be implemented there for Matter BTP, as the [Matter BTP implementation](https://github.com/ivmarkov/rs-matter-embassy/blob/79a2a7786ad28e2ae186e4136e22c93a2c343599/rs-matter-embassy/src/ble.rs) is now a hybrid between processing some PDUs using the [low-level now-public `trouble::att` module](https://github.com/ivmarkov/rs-matter-embassy/blob/79a2a7786ad28e2ae186e4136e22c93a2c343599/rs-matter-embassy/src/ble.rs#L313), and also delegating the "metadata" types of queries to a [custom Matter BTP-specific GATT server](https://github.com/ivmarkov/rs-matter-embassy/blob/79a2a7786ad28e2ae186e4136e22c93a2c343599/rs-matter-embassy/src/ble.rs#L59), implemented with the proc-macro that got merged a couple of months ago.

Other than that, I must say I'm relying on the GATT proc-macro-generated server to be possible to instantiate multiple times (see #250 why this is not possible in the general case).

Miraculously, it does work for me - even in the presence of the `static_cell` thing, but my suspicion is it might work simply because I use [0-length `[u8; 0]` payload for the C1 and C2 characteristics](https://github.com/ivmarkov/rs-matter-embassy/blob/79a2a7786ad28e2ae186e4136e22c93a2c343599/rs-matter-embassy/src/ble.rs#L49) in the above proc-macro-generated GATT server so as to conserve memory. While these are not really 0-length, the trick is that no "Read" or "Write" or "Indicate" for those is reaching the GATT server simply because I handle the "Read"/"Write"/"Indicate" on the ATT layer level (hence the "hybrid" approach).

closed 2025-02-05 by ivmarkov