embassy-rs/trouble #136: Consolidate methods and data onto characteristics

petekubiak, 2024-10-10:
I believe characteristic actions can be simplified by adding methods and persistent data to each characteristic instead of requiring access through the AttributeTable. This has the potential to remove a layer of complexity from the library by removing the need for the AttributeServer and AttributeTable.

This follows on from the feature developed in #131 to define GATT services and characteristics through attribute procedural macros. The idea is to implement a model similar to nrf_softdevice where actions such as `get()`, `set()` and `notify()` are tied to the characteristics themselves rather than being called on the GattServer and AttributeTable types.

This work is closely linked with #135 as both require changes to the same area of the code. I will attempt to put together a draft propsal for this so that we can discuss further how this can be implemented in such a way that it incorporates the Matter BTP requirements outlined in #135.

alexmoon, 2024-10-10:
That sounds like a pretty good idea to me. I think that should apply generally to Attributes (e.g. either characteristics or descriptors). I also think that they should come in two flavors: host managed or app managed. With a host-managed attribute, the host has the persistent data and updates or it when receiving requests from the remote. With an app-managed attribute, the host uses events/callbacks/whatever to notify the app of remote requests and the app is responsible for providing the response.

lulf, 2024-10-14:
Agreed it would be nice to use the characteristics directly for this.

closed 2024-11-13 by petekubiak