embassy-rs/trouble #149: Add support for variable length charactersitics

petekubiak, 2024-10-25:
Currently, characteristic data is treated as a fixed size byte array. This means that characteristics with variable length data are not fully supported.

If a characteristic is defined as a `heapless::Vec`, the entire struct will be sent (metadata and payload) regardless of the current length of the data inside. On receiving, it is likely that a write command will be rejected unless a byte array matching the exact size of the `heapless::Vec` in memory is received.

nrf-softdevice interprets characteristics with the Vec type as variable length, and will only send bytes corresponding to the current length of the vector on transmission. On reception, it will read in a variable number of bytes and update the length an contents of the characteristic's vector accordingly.

I think it's worth implementing the same functionality in TrouBLE.

lulf, 2024-10-25:
I think #133 would also allow this maybe in a more efficient way?

petekubiak, 2024-10-28:
Ah interesting. In all honesty when I first read #133 I didn't have enough knowledge of TrouBLE to properly absorb the extent of the changes. I'll give it another look now I've had a few more weeks' exposure.

closed 2025-01-07 by lulf