embassy-rs/trouble #152: Add a way to validate/interact with a value before it gets put in AttributeTable

Erik1000, 2024-10-30:
Currently, you can only passively monitor the GattEvent. This is problematic in the case of `GattEvent::Write` because there is no way you can validate or inspect the value of the write event before it gets put into the AttributeTable and can then be read again.
It would be nice to hook into the write event to validate the value and (in case of error) discard it.

Erik1000, 2024-10-30:
I think in general it would be nice to be able to handle the actual values read and written better. For example, manually hook into reads and writes and controlling which value is sent or stored.

jamessizeland, 2024-10-30:
@petekubiak has been looking at some of this, we're talking about some kind of middleware layer with callbacks essentially right?

petekubiak, 2024-10-30:
I think this links closely with #135 - a draft PR (#133) is open to suggest ideas for how this might be achieved.

petekubiak, 2024-11-07:
#151 is now up for review - I would be interested to hear your thoughts on the approach

petekubiak, 2024-11-18:
@Erik1000 are you happy for this issue to be closed?

Erik1000, 2024-11-18:
Yes

closed 2024-11-18 by Erik1000