embassy-rs/trouble #157: Scan feature breaks build

pierrickrouxel, 2024-11-08:
When I build the project, rust throws the attached error.
I'm pretty new in the rust community, so I not able to understand why it doesn't work.

The error line is:

```rs
central.scan(&ScanConfig::default());
```

The library is imported with:

```toml
trouble-host = { git = "https://github.com/embassy-rs/trouble", package = "trouble-host", features = [
    "log",
    "scan",
] }
```

<details>
<summary>Build log</summary>

```
error[E0277]: the trait bound `C: ControllerCmdSync<LeSetScanParams>` is not satisfied
   --> src/ble_bas_central.rs:36:17
    |
36  |         central.scan(&ScanConfig::default());
    |                 ^^^^ the trait `ControllerCmdSync<LeSetScanParams>` is not implemented for `C`
    |
note: required by a bound in `trouble_host::Central::<'d, C>::scan`
   --> trouble-148cf413cadab653/3a9d3c1/host/src/central.rs:290:12
    |
288 |     pub async fn scan(&mut self, config: &ScanConfig<'_>) -> Result<ScanReport, BleHostError<C::Error>>
    |                  ---- required by a bound in this associated function
289 |     where
290 |         C: ControllerCmdSync<LeSetScanParams>
    |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ required by this bound in `Central::<'d, C>::scan`
help: consider further restricting this bound
    |
20  |     C: Controller + bt_hci::controller::ControllerCmdSync<bt_hci::cmd::le::LeSetScanParams>,
    |                   +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

error[E0277]: the trait bound `C: ControllerCmdSync<LeSetScanEnable>` is not satisfied
   --> src/ble_bas_central.rs:36:17
    |
36  |         central.scan(&ScanConfig::default());
    |                 ^^^^ the trait `ControllerCmdSync<LeSetScanEnable>` is not implemented for `C`
    |
note: required by a bound in `trouble_host::Central::<'d, C>::scan`
   --> trouble-148cf413cadab653/3a9d3c1/host/src/central.rs:291:15
    |
288 |     pub async fn scan(&mut self, config: &ScanConfig<'_>) -> Result<ScanReport, BleHostError<C::Error>>
    |                  ---- required by a bound in this associated function
...
291 |             + ControllerCmdSync<LeSetScanEnable>
    |               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ required by this bound in `Central::<'d, C>::scan`
help: consider further restricting this bound
    |
20  |     C: Controller + bt_hci::controller::ControllerCmdSync<bt_hci::cmd::le::LeSetScanEnable>,
    |                   +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
```

</details>

lulf, 2024-11-08:
Which controller implementation are you using with trouble-host? Could you share some more of the initialization code, as well as which target you are compiling against.

pierrickrouxel, 2024-11-08:
Thank you for your quick answer. I'm on esp32h2.

I reassembled the code from 2 classes for the exemple:

```rs
#![no_std]
#![no_main]

use bt_hci::controller::ExternalController;
use embassy_executor::Spawner;
use esp_alloc as _;
use esp_backtrace as _;
use esp_hal::{prelude::*, timer::timg::TimerGroup};
use esp_wifi::ble::controller::asynch::BleConnector;

#[esp_hal_embassy::main]
async fn main(_s: Spawner) {
    esp_println::logger::init_logger_from_env();
    let peripherals = esp_hal::init({
        let mut config = esp_hal::Config::default();
        config.cpu_clock = CpuClock::max();
        config
    });

    esp_alloc::heap_allocator!(72 * 1024);

    let timg0 = TimerGroup::new(peripherals.TIMG0);

    let init = esp_wifi::init(
        esp_wifi::EspWifiInitFor::Ble,
        timg0.timer0,
        esp_hal::rng::Rng::new(peripherals.RNG),
        peripherals.RADIO_CLK,
    )
    .unwrap();

    let systimer = esp_hal::timer::systimer::SystemTimer::new(peripherals.SYSTIMER)
        .split::<esp_hal::timer::systimer::Target>();
    esp_hal_embassy::init(systimer.alarm0);

    let mut bluetooth = peripherals.BT;
    let connector = BleConnector::new(&init, &mut bluetooth);
    let controller: ExternalController<_, 20> = ExternalController::new(connector);

    let mut resources = Resources::new(PacketQos::None);
    let (stack, _, mut central, mut runner) = trouble_host::new(controller, &mut resources).build();

    info!("Scanning for peripheral...");
    let _ = join(runner.run(), async {
        info!("Connecting");

        central.scan(&ScanConfig::default()); // Error is here
    })
    .await;
}
```

lulf, 2024-11-08:
This might be an issue with the `bt-hci` version. Make sure you have a dependency on version `0.1.1`. Also, I've confirmed it works with esp-hal revision `208339ddeb5b0747d1403d611d9c8718f5bf4f08` and trouble rev `9840438d75f0bfcc0cf950f5d1b267e86736b7b3`, make sure that you use those revs.

pierrickrouxel, 2024-11-08:
Not better. You can see my Cargo.toml after:

```toml
[package]
name = "hello_esp_nostd"
version = "0.1.0"
authors = ["Pierrick Rouxel"]
edition = "2021"
license = "MIT"

[dependencies]
embassy-executor = { version = "0.6.1", features = [
    "integrated-timers",
    "arch-riscv32",
] }
embassy-time = { version = "0.3.2" }
trouble-host = { git = "https://github.com/embassy-rs/trouble", package = "trouble-host", features = [
    "log",
    "scan",
] }
esp-backtrace = { version = "0.14.2", features = [
    "panic-handler",
    "exception-handler",
    "esp32h2",
    "println",
] }
esp-hal = { version = "0.21.1", features = ["esp32h2"] }
esp-hal-embassy = { version = "0.4.0", features = [
    "esp32h2",
    "integrated-timers",
] }
esp-println = { version = "0.12.0", features = ["log", "esp32h2"] }
log = "0.4.22"
esp-wifi = { version = "0.10.1", features = ["async", "ble", "esp32h2"] }
embassy-futures = { version = "0.1.1", features = ["log"] }
esp-alloc = "0.5.0"
bt-hci = { version = "0.1.1", features = ["embassy-time", "log"] }
embassy-sync = { version = "0.6.0", features = ["log"] }

[profile.dev]
# Rust debug is too slow.
# For debug builds always builds with some optimization
opt-level = "s"

[profile.release]
codegen-units = 1        # LLVM can perform better optimizations using a single thread
debug = 2
debug-assertions = false
incremental = false
lto = 'fat'
opt-level = 's'
overflow-checks = false

[patch.crates-io]
esp-hal = { git = "https://github.com/esp-rs/esp-hal.git", rev = "208339ddeb5b0747d1403d611d9c8718f5bf4f08" }
esp-hal-embassy = { git = "https://github.com/esp-rs/esp-hal.git", rev = "208339ddeb5b0747d1403d611d9c8718f5bf4f08" }
esp-wifi = { git = "https://github.com/esp-rs/esp-hal.git", rev = "208339ddeb5b0747d1403d611d9c8718f5bf4f08" }
esp-println = { git = "https://github.com/esp-rs/esp-hal.git", rev = "208339ddeb5b0747d1403d611d9c8718f5bf4f08" }
esp-backtrace = { git = "https://github.com/esp-rs/esp-hal.git", rev = "208339ddeb5b0747d1403d611d9c8718f5bf4f08" }
esp-alloc = { git = "https://github.com/esp-rs/esp-hal.git", rev = "208339ddeb5b0747d1403d611d9c8718f5bf4f08" }
```

lulf, 2024-11-08:
> Not better. You can see my Cargo.toml after:
>
> ```toml
> [package]
> name = "hello_esp_nostd"
> version = "0.1.0"
> authors = ["Pierrick Rouxel"]
> edition = "2021"
> license = "MIT"
>
> [dependencies] [...]

Try changing this

> trouble-host = { git = "https://github.com/embassy-rs/trouble", package = "trouble-host", features = [
>     "log",
>     "scan",
> ] }

To this

`trouble-host = { version = "0.1.0", features = [ "log", "scan"] }`

> esp-backtrace = { version = "0.14.2", features = [
>     "panic-handler",
>     "exception-handler",
>     "esp32h2",
>     "println",
> ] }
> esp-hal = { version = "0.21.1", features = ["esp32h2"] } [...]

And add this to the patch section:

`trouble-host = { git = "https://github.com/embassy-rs/trouble.git", rev = "9840438d75f0bfcc0cf950f5d1b267e86736b7b3" }`

pierrickrouxel, 2024-11-08:
Humpf, I think it's my code that contains a mistake. If I merge my code in one file it's ok.

I spliced the code in two modules but my Rust understanding seems to be bad. I think it missing things in my where clause.
I will read more :)

Sorry. 

My module:
```rs
use embassy_futures::join::join;
use esp_alloc as _;
use esp_backtrace as _;
use log::info;
use trouble_host::{scan::ScanConfig, Controller, HostResources, PacketQos};

/// Size of L2CAP packets
const L2CAP_MTU: usize = 128;

/// Max number of connections
const CONNECTIONS_MAX: usize = 1;

/// Max number of L2CAP channels.
const L2CAP_CHANNELS_MAX: usize = 3; // Signal + att + CoC

type Resources<C> = HostResources<C, CONNECTIONS_MAX, L2CAP_CHANNELS_MAX, L2CAP_MTU>;

pub async fn run<C>(controller: C)
where
    C: Controller,
{
    let mut resources = Resources::new(PacketQos::None);
    let (stack, _, mut central, mut runner) = trouble_host::new(controller, &mut resources).build();

    info!("Scanning for peripheral...");
    let _ = join(runner.run(), async {
        info!("Connecting");

        let reports = central.scan(&ScanConfig::default()).await.unwrap();
    })
    .await;
}
```

closed 2024-11-08 by pierrickrouxel