embassy-rs/trouble #168: async is_connected

jamessizeland, 2024-11-14:
## Issue:

A common pattern we are using some variation of the following:

```rust
/// A BLE GATT server
pub async fn gatt_server_task(server: &Server<'_>, conn: &Connection<'_>) {
    // check if the connection is still active every second
    loop {
        match select(Timer::after_secs(1), server.next()).await {
            Either::First(_) => {
                if !conn.is_connected() {
                    break;
                }
            }
            Either::Second(event) => match event {
                ...
            },
        }
    }
    info!("Gatt server task finished, return to advertising");
}

#[embassy_executor::main]
async fn main(spawner: Spawner) {
    ...

    // Main loop
    loop {
        // advertise for connections
        info!("advertising for connection");
        if let Ok(conn) = advertiser.advertise().await {
            info!("connected!");

            let sensor1 = sensor1_task(&server, &conn);
            let sensor2 = sensor2_task(&server, &conn);
            let sensors_fut = embassy_futures::select::select(sensor1, sensor2);
            let gatt_fut = gatt_server_task(&server, &conn);
            embassy_futures::select::select(gatt_fut, sensors_fut).await; // sensors tasks dropped when connection closes
        }
    }
```

Where the gatt_server_task is used as a monitor for the lifecycle of the gatt connection.

## Proposal

I think it would be ergonomic to not have to poll for `conn.is_connected()`.  This probably has an impact on power consumption? Rather  to have this disconnect event come through as another event on server.next() or have an async function something like `conn.connection_ended().await` that can be used instead to close the gatt_server_task.

Does anyone have a preference on the way this should be implemented?

petekubiak, 2024-11-14:
We had been chatting about two possible avenues for this:
- Add a `GattEvent::Disconnection(Connection)` variant to allow user to handle disconnection
- Implement a `Stream` which terminates to indicate disconnection

I'm leaning towards the `Disconnection` variant approach as the server can handle multiple connections concurrently, which would then either require multiple streams or a stream which keeps track of all connections and only terminates once all connections are closed, which doesn't sound particularly intuitive.

alexmoon, 2024-11-14:
I agree that a disconnection event is better because the server outlives the connection. It should also include the disconnect reason from the GAP packet.

I could see an alternate design such that the stream of events is associated with the connection rather than the server. So you'd get events something like `ConnectionEvent::GattWrite(server, data)` instead of `GattEvent::Write(connection, data)`. In that model you would terminate the stream after disconnect. An advantage of that structure is that it would be very easy to have one task per connection handling all of the connection-oriented events (L2CAP, GATT, etc). However, you'd probably still want a `Disconnection` event so that you can indicate the disconnect reason to the application.

lulf, 2024-11-15:
I've had a stab at supporting connection events here: https://github.com/embassy-rs/trouble/pull/170 

I'm thinking that can be extended to support GattEvents as well, with a feature gate on "gatt". One thing to keep in mind though is the lifetime of the gatt write data, which I think need to somehow be forwarded from the GattServer next() to the connection event queue (and in the process, perhaps that should be renamed run() and loop forever while dispatching gatt events to the connections?).

jamessizeland, 2024-11-15:
Nice speedy work! I'll pull it into our project and give it a try 🙂.

closed 2024-11-18 by jamessizeland