embassy-rs/trouble #175: Add default_value and descriptor to characteristic macro

HaoboGu, 2024-11-22:
Current gatt service macro fills 0s as the `AttributeData`:

https://github.com/embassy-rs/trouble/blob/224f37194d593cd884f736c0f168b097f9b61a5d/host-macros/src/service.rs#L149

It would be very useful if users could set the value of `AttributeData`.

I found the following code blocks in `host-macro`, it seems `value` and `descriptor` macro args are not implemented. May I ask that is there any plan to finish this?

https://github.com/embassy-rs/trouble/blob/224f37194d593cd884f736c0f168b097f9b61a5d/host-macros/src/characteristic.rs#L115-L126

jamessizeland, 2024-11-22:
It is on the to-do list yes 😊, you're welcome to PR an implementation, but also good to know it's a wanted feature so we can prioritise it.

HaoboGu, 2024-11-23:
yeah, this is commonly used in HID over BLE, for example, setting hid  info and report map

lulf, 2024-12-09:
Agreed, I think that could be useful for the device information service as well.

jamessizeland, 2024-12-09:
Do we have a preference on the interface for descriptors? I can't work out how they are expected to look in Softdevice but I was wondering if they should be also part of the #[characteristic(...)] or if we should also have #[descriptor(...)] too and allow multiple of them per characteristic.

@petekubiak  and I have got the default value stuff working with nice type checking and error messages, I'll open a draft now to show that.

lulf, 2024-12-09:
It should be possible to use multiple descriptors per characteristic (in fact someone is already doing that, see https://github.com/embassy-rs/trouble/issues/95)

jamessizeland, 2024-12-09:
cool, I'll take a look at that, I hadn't spotted thanks.

jamessizeland, 2024-12-09:
ah sorry on the multiple descriptors front I meant either something like:

```rust
// Battery service
#[gatt_service(uuid = "180f")]
struct BatteryService {
    #[characteristic(uuid = "2a19", read, write, notify, descriptior = (uuid = "1234", value = "something"), descriptor = (uuid = "2345", value = "something else"))]
    level: u8,
}
```

or

```rust
// Battery service
#[gatt_service(uuid = "180f")]
struct BatteryService {
    #[characteristic(uuid = "2a19", read, write, notify)]
    #[descriptor(uuid = "1234", value = "something")]
    #[descriptor(uuid = "2345", value = "something else")]
    level: u8,
}

```

lulf, 2024-12-09:
> ah sorry on the multiple descriptors front I meant either something like:
>
> ```rust
> // Battery service
> #[gatt_service(uuid = "180f")]
> struct BatteryService {
>     #[characteristic(uuid = "2a19", read, [...]

This variant gets my vote, I think it is more readable.

HaoboGu, 2024-12-10:
i prefer the second, too

jamessizeland, 2024-12-10:
I've got docstrings parsing and being put back into the generated gatt_service structs, but for some reason it doesn't show up in rust analyzer :/

lulf, 2025-01-07:
This is done right?

closed 2025-01-07 by lulf