embassy-rs/trouble #178: Compile-time validation for Advertisting data

plaes, 2024-11-25:
It's currently quite easy to create a compilable and runnable example code, but which doesn't show up in the nrf connect:
```rust
    let mut adv_data = [0; 31];
    AdStructure::encode_slice(
        &[
            AdStructure::Flags(LE_GENERAL_DISCOVERABLE | BR_EDR_NOT_SUPPORTED),
            AdStructure::ServiceUuids16(&[Uuid::Uuid16([0x0f, 0x18])]),
            AdStructure::CompleteLocalName(b"12345678901234567890123"), // Causes silent advertising failure
            // AdStructure::CompleteLocalName(b"1234567890123456789012"), // WORKS
        ],
        &mut adv_data[..],
    )?;
// ...
```

jamessizeland, 2024-11-25:
This will be because of the packet size limit of 22 bytes for name. I've put a check in my latest PR, but I agree it could be a built in assertion.

```rust
    let name = if name.len() > 22 { 
         let truncated_name = &name[..22]; 
         info!("Name truncated to {}", truncated_name); 
         truncated_name 
     } else { 
         name 
     };
```

plaes, 2024-11-25:
There's something else as well, because device does not appear when scanning with NRF connect on Android,

lulf, 2024-11-29:
It will not advertise, because the advertise_task() will fail. Rather than truncating, I think it's better to panic on this error, I've raised a PR for this https://github.com/embassy-rs/trouble/pull/187

closed 2024-11-29 by lulf