embassy-rs/trouble #180: Service Changed - 0x2A05

saltedpotatos, 2024-11-25:
Is there any documentation on how to setup the Service Changed characteristic? I've found this [Stack Overflow](https://stackoverflow.com/questions/73514239/how-do-i-use-the-service-changed-characteristic-to-invalidate-attributes-cache) post, but I haven't been able to make that work. 

It would be nice to not have to turn off bluetooth and reboot my phone to clear the cache every time I flash any changes involving BLE characterstics to the micro controller

lulf, 2024-12-09:
Could you share what modifications you've tried? If you use the macro API, you have to set the characteristic value after creating the handle. If you use the builder api, you can use add_characteristic_ro with the value (handle range)

saltedpotatos, 2024-12-09:
So perhaps I don't understand the value I need to create,  but I can certainly show what doesn't work! 

So I've got a gatt_server, which also sets up a generic service 0x1801, which should hold the service_changed characteristic. 

```rust
// GATT Server definition
#[gatt_server]
pub struct Server {
    pub generic: Generic,
    pub fault_service: FaultIdentifierService,
    pub relay_service: RelayService,
}

#[gatt_service(uuid = "1801")]
pub(crate) struct Generic {
    #[characteristic(uuid = "2A05", indicate, read)]
    service_changed: i32,
}

```

and then I can try and set the value for the service changed. 

```rust
        //Note: A server may set the affected Attribute Handle range to 0x0001 to 0xFFFF to indicate to the client to rediscover the entire set of Attribute Handles on the server.
        let handle_start = 0x0001;
        let handle_end = 0xFFFF;
        let full_value = handle_start + handle_end;
        let test = server.set(&server.generic.service_changed, &full_value);
        info!("We were able to set service changed? {test:?}");

```

But scanning with NRF Connect, I can see the 0x1801 service, but it does not contain the service_changed characteristic.

saltedpotatos, 2024-12-10:
Reading some [docs](https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Core-54/out/en/host/generic-attribute-profile--gatt-.html#UUID-542d3f08-c238-c464-d662-b3982828e5bc) , I've also tried concatenating the handle start & end, as well as having the characteristic be indicate only. I have not noticed any change though 
```rust
#[gatt_service(uuid = "1801")]
pub(crate) struct Generic {
    #[characteristic(uuid = "2A05", indicate)]
    service_changed: u32,
}

        let handle_start = 0x0001_u16;
        let handle_end = 0xFFFF_u16;
        let concatenated: u32 = ((handle_start as u32) << 16) | handle_end as u32;
        info!("Concatenated: {concatenated:?}");
        let test = server.set(&server.generic.service_changed, &concatenated);
        info!("We were able to set service changed? {test:?}");

```

lulf, 2024-12-11:
So, I believe each handle is little endian, and it should be:

```
    let start = 0x0001;
    let end = 0xFFFF;
    
    let mut value: [u8; 4] = [0; 4];
    value[0..4].copy_from_slice(&start.to_le_bytes());
    value[4..8].copy_from_slice(&end.to_le_bytes());
```

I think there might be another issue as well, namely that the macros will add the 1801 generic service automatically, and you might end up with 2 of that same service. I think you'd need to use the builder api for now to work around that. We should ideally modify the macros to allow adding optional characteristics to it somehow. 

CC @jamessizeland

saltedpotatos, 2024-12-19:
Alright, I believe I have the world's simplest POC for clearing the service cache.

It still does not list the service changed characteristic in nRF Connect, but it rediscovers the services / characteristics on every connect, so that's good enough to keep the ball rolling. Thanks for the help @lulf !

```
diff --git a/host/src/gap.rs b/host/src/gap.rs
index a0f0720..5f666d8 100644
--- a/host/src/gap.rs
+++ b/host/src/gap.rs
@@ -93,7 +93,9 @@ impl<'a> PeripheralConfig<'a> {
         gap_builder.add_characteristic_ro(characteristic::APPEARANCE, self.appearance);
         gap_builder.build();
 
-        table.add_service(Service::new(service::GATT));
+        let mut gatt_builder = table.add_service(Service::new(service::GATT));
+
+        gatt_builder.add_characteristic_ro(characteristic::SERVICE_CHANGED, &[1, 0, 255, 255]);
 
         Ok(())
     }

```

This will require you to set a custom attribute_table_size
```rust
#[gatt_server(attribute_table_size = 50)]
```

or it will crash on first boot after flash with
`trouble/host/src/attribute.rs:310:13:
no space for more attributes`

If I understood the macro syntax better (or at all), I could probably get that to include our added characteristic in the table size calculation during compile time, but that's a problem for NOT the world's simplest POC.

lulf, 2025-01-07:
Closing this for now, thanks for the report.

closed 2025-01-07 by lulf