embassy-rs/trouble #197: Newcomer question

akauppi, 2024-12-16:
I have my custom service running, and can interact with it using nRF Connect application.

Defined as:

```rust
#[gatt_service(uuid = "438785e7-4942-4749-a072-dceb73fd6c87")]
struct MyService {
    #[characteristic(uuid = "448785e7-4942-4749-a072-dceb73fd6c87", write, on_write = rgb_on_write)]
    rgb: [u8;3],
    #[characteristic(uuid = "458785e7-4942-4749-a072-dceb73fd6c87", read, notify, on_read = magic_on_read)]
    magic: u32
}
```

How can I give names to the service and characteristics? 

[image]

Do the names matter, in practical situations; i.e. am I "just fine" with them being Unknowns?

jamessizeland, 2024-12-16:
Unknown here means that the UUIDs you have chosen aren't from the list of predefined Characteristics from the Bluetooth Special Interest Group. That's totally fine, most UUIDs are 'custom' like you're using there. The predefined ones are written in here if you want to try them out https://github.com/embassy-rs/bt-hci/blob/main/src%2Fuuid%2Fcharacteristic.rs

lure23, 2024-12-16:
Thanks for the answer. Happy to leave them as-is, now that I understand it's not due to my lack of some setup.

>the BluetoothUuid16 type can't currently be used in the macro though, so I'll have to look into a neat way to resolve that.

Is that.. about the characteristics also having 128-bit UUID's?  Or the description?

The usage is adapted from `bleps` sample. If it's possible for the characteristics to have Uuid16, I'd probably do that. But not even a speed bump for me.

Ready to close the issue - I got the answers... do you still wish to keep this open / refocus it?

jamessizeland, 2024-12-16:
Happy to close. You can currently also use 16bit UUIDs but you have to use them as strings, like you are now. I'd want us to also allow using the constants from bt-hci which are all BluetoothUuid16(u16)

closed 2024-12-16 by akauppi