embassy-rs/trouble #206: UUIDs from Into<Uuid>

jamessizeland, 2024-12-31:
Currently we can only accept UUIDs (for characteristics and services in the macro) as string literals.

We now have a set of constants from bt-hci that could be used for known UUIDs, but these aren't accepted by the macro.

Looking at making the uuid input accept anything that impl Into<Uuid> and as a by product I'm wondering about if we should switch out our custom Uuid for the crate one.

lure23, 2024-12-31:
Hugely interested in this, since I would like to bring my custom UUIDs as consts. Unfortunately, not up to editing macros.

jamessizeland, 2025-01-03:
https://github.com/jamessizeland/trouble

I'm working on it here if you want to see. 

Macros take some getting used to. But cargo-expand is your friend there for seeing what a macro builds under the hood. 

Feel free to ask questions about any of it.

lure23, 2025-01-03:
> Macros take some getting used to. But cargo-expand is your friend there for seeing what a macro builds under the hood.

Sure! I tried it but - at least for me - got the idea it's mainly useful when the code base builds. For a broken build (somewhere else than in a macro), I don't have a way to see what the output would be.  ((Eventually, I hope IDE's would fix this; have filed a FR to Rust Rover))

> Feel free to ask questions about any of it.

Thanks!  This is off-topic so let's keep the non-thread short. :)

closed 2025-01-07 by lulf