embassy-rs/trouble #209: RSSI

Makuo12, 2024-12-31:
Sorry I wanted to ask if there is any implementation of RSSI in trouble. If so please could I see like a short implementation or a direction to look at.

lulf, 2025-01-01:
The stack is basically a clonable copyable copy of a reference to the Host and Controller, and when you pass this to the connection.rssi(), it will query the controller for the RSSI information for that client.

More information about the RSSI format is here: https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Core-54/out/en/host-controller-interface/host-controller-interface-functional-specification.html#UUID-97d75ef1-ae03-1164-a6be-61c65dc9fb94

[image]

Makuo12, 2025-01-01:
Thank you @lulf. I believe I understand what the stack is and how to get it now. 
```
pub async fn run<C>(controller: C)
where
    C: Controller + ControllerCmdSync<ReadRssi>,
{
    let mut resources = Resources::new(PacketQos::None);
    let (stack, mut peripheral, _, runner) = trouble_host::new(controller, &mut resources)
        .set_random_address(address)
        .build();
    info!("Starting advertising and GATT service");
    let server = Server::new_with_config(GapConfig::Peripheral(PeripheralConfig {
        name: "XOX",
        appearance: &appearance::network_device::GENERIC_NETWORK_DEVICE,
    }))
    .unwrap();
    let app_task = async {
        loop {
            match advertise("XOX", &mut peripheral).await {
                Ok(conn) => {
                    info!("rssi now {}", conn.rssi(stack).await.unwrap());
```

Makuo12, 2025-01-01:
When I call connection.disconnect() I get a panic error. So I tried using next(), which made it start listening for the next connection. I want to ask does disconnect remove the BleHost and does next() release the current connection to listen for the another incoming connection.
```
                Ok(conn) => {
                    delay.delay_millis(2000);
                    let value = conn.rssi(stack.clone()).await.unwrap();
                    info!("value {}", value);
                    if value > -20 {
                        // set up tasks when the connection is established to a central, so they don't run when no one is connected.
                        let connection_task = conn_task(&server, &conn);
                        let counter_task = counter_task(&server, &conn);
                        // run until any task ends (usually because the connection has been closed),
                        // then return to advertising state.
                        // connection_task.await;
                        select(connection_task, counter_task).await;
                    } else {
                         conn.disconnect();
                        // conn.next().await;
                    }
```

lulf, 2025-01-01:
Where does it panic?

Makuo12, 2025-01-01:
```
            match advertise("XOX", &mut peripheral).await {
                Ok(conn) => {
                    delay.delay_millis(2000);
                    let value = conn.rssi(stack.clone()).await.unwrap();
                    info!("value {}", value);
                    if value > -20 {
                        // set up tasks when the connection is established to a central, so they don't run when no one is connected.
                        let connection_task = conn_task(&server, &conn);
                        let counter_task = counter_task(&server, &conn);
                        // run until any task ends (usually because the connection has been closed),
                        // then return to advertising state.
                        // connection_task.await;
                        select(connection_task, counter_task).await;
                    } else {
                        conn.disconnect();
                        // conn.next().await;
                    }
                    
                }
                Err(e) => {
                    #[cfg(feature = "defmt")]
                    let e = defmt::Debug2Format(&e);
                    panic!("[adv] error: {:?}", e);
                }
            }
```
```
async fn ble_task<C: Controller>(mut runner: Runner<'_, C>) {
    loop {
        if let Err(e) = runner.run().await {
            #[cfg(feature = "defmt")]
            let e = defmt::Debug2Format(&e);
            panic!("[ble_task] error: {:?}", e);
        }
    }
}
```
When I call the conn.disconnect(); it panics from the ble_task. With the error [ble_task] error: BleHost(NotFound)

jamessizeland, 2025-01-04:
@Makuo12 what hardware are you using for this testing?

jamessizeland, 2025-01-04:
https://github.com/embassy-rs/trouble/pull/219 I've put this together to try to debug your issue.  Seems to be working just fine on esp32c3, which is all I've got to hand right now.

Makuo12, 2025-01-04:
@jamessizeland i am using a normal esp32.

Makuo12, 2025-01-04:
@jamessizeland thank you for the example, it works fine. I also wanted to ask if you could pls take a look at #210. I dont know if the issue is from my esp32 microcontroller. When I use the descriptor, characteristic it doesn't work as I intend.

Makuo12, 2025-01-06:
@jamessizeland just saw the changes made with commit 5e55ac1. Thank you it now works fine for me.

closed 2025-01-06 by Makuo12