embassy-rs/trouble #214: Confused about state of `on_write`

lure23, 2025-01-03:
https://github.com/embassy-rs/trouble/blob/main/host-macros/src/characteristic.rs#L190-L191

`on_write` is commented out, but still mentioned in docs. What's going on?

I'm trying to update https://github.com/jamessizeland/microbit-ble-gamepad to latest trouble, and ... just understand where things are, at the moment. And where they are going. Please advice some

lure23, 2025-01-03:
Also: is
>#[gatt_server(attribute_table_size = 100)]

the same as earlier
>#[gatt_server(attribute_data_size = 100)]

jamessizeland, 2025-01-03:
Hi, we're moving away from the callbacks to using the server.next() interface instead and improving that API. The gamepad repo is a little bit behind the new activity that's been done on Trouble over the holidays. We'll need to update the microbit-bsp to latest TrouBLE and then the gamepad project to match.

I can do that over the weekend if you like, unless you're keen to go through the process yourself. Feel free to open issues there for me to respond to.

As for the table size attribute, this is now automatically created in the latest versions of TrouBLE, so you don't need to specify it anymore.

akauppi, 2025-01-03:
>I can do that over the weekend if you like, unless you're keen to go through the process yourself. Feel free to open issues there for me to respond to.

I've started going through that, but if you're anyways going to update, it might be better for me to wait.

>As for the table size attribute, this is now automatically created in the latest versions of TrouBLE, so you don't need to specify it anymore.

Got it. All this really appreciated!!! :)

lure23, 2025-01-03:
>I can do that over the weekend if you like, unless you're keen to go through the process yourself. Feel free to open issues there for me to respond to.

I've started going through that, but if you're anyways going to update, it might be better for me to wait.

>As for the table size attribute, this is now automatically created in the latest versions of TrouBLE, so you don't need to specify it anymore.

Got it. All this really appreciated!!! :)

closed 2025-01-03 by lure23