embassy-rs/trouble #229: Panic on Peripheral::advertise when using esp32c6

eloycoto, 2025-01-09:
When the reaching here:

https://github.com/embassy-rs/trouble/blob/419e29886ee696570f0513b76b7aa52f82a19ff5/host/src/peripheral.rs#L64-L73

the BleHost:command is called, but if fails on OnceLock on initalized

```rust
    pub(crate) async fn command<C>(&self, cmd: C) -> Result<C::Return, BleHostError<T::Error>>
    where
        C: SyncCmd,
        T: ControllerCmdSync<C>,
    {
        let _ = self.initialized.get().await;

        let ret = cmd.exec(&self.controller).await?;
        Ok(ret)
    }
```

And the panic:

```
INFO - Our address = Address { kind: AddrKind(1), addr: BdAddr([65, 90, 227, 30, 131, 231]) }
INFO - Starting advertising and GATT service
INFO - [host] filter accept list size: 12
INFO - [host] setting txq to 24, fragmenting at 255
INFO - [host] configuring host buffers (8 packets of size 251)

====================== PANIC ======================
panicked at /tmp/trouble_bk/examples/apps/src/ble_bas_peripheral.rs:104:13:
[ble_task] error: BleHost(Hci(Invalid HCI Command Parameters))

Backtrace:

0x4200e91e
0x4200e91e - embassy_executor::raw::util::SyncUnsafeCell<T>::get
    at /home/eloy/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-executor-0.7.0/src/raw/util.rs:55
0x42016b90
0x42016b90 - main
    at /tmp/trouble_bk/examples/esp32/src/bin/ble_bas_peripheral.rs:11
0x420055e8
0x420055e8 - hal_main
    at /home/eloy/.cargo/git/checkouts/esp-hal-42ec44e8c6943228/7288d03/esp-hal/src/lib.rs:485
```

I might fix it, but I need to check how to use gdb with espflash, etc.. it might take a while, so any hint it's welcome :-)

eloycoto, 2025-01-10:
Ok, found the problem! host is not initialized, so advertise breaks, need to wait until 

https://github.com/embassy-rs/trouble/blob/419e29886ee696570f0513b76b7aa52f82a19ff5/host/src/host.rs#L831-L833

lulf, 2025-01-10:
@eloycoto Is this command producing an error or just hanging?

lure23, 2025-01-10:
> @eloycoto Is this command producing an error or just hanging?

It panics, as posted by @eloycoto :

>```
>====================== PANIC ======================
>panicked at /tmp/trouble_bk/examples/apps/src/ble_bas_peripheral.rs:104:13:
>[ble_task] error: BleHost(Hci(Invalid HCI Command Parameters))
>```

eloycoto, 2025-01-10:
> @eloycoto Is this command producing an error or just hanging?

In this case, it's not panic (It panics in the example), just return an invalid parameter error.

eloycoto, 2025-01-10:
And the params are the following:

```
configuring host buffers (8 packets of size 251)
```

And this is the error:

```
BleHost(Hci(Invalid HCI Command Parameters))
```

eloycoto, 2025-01-10:
Was checking the doc for the command:

https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Core-54/out/en/host-controller-interface/host-controller-interface-functional-specification.html#UUID-99748527-de87-bf9c-935d-baf7e3f35b12

And looks like that `Host_Synchronous_Data_Packet_Length` should be 1, 0 is not a valid one:

Tried this, and also fails :-( 

```
 HostBufferSize::new( 
     host.rx_pool.mtu() as u16, 
     1, 
     config::L2CAP_RX_PACKET_POOL_SIZE as u16, 
     0, 
 ) 
 .exec(&host.controller) 
 .await?; 
 ```

sjroe, 2025-01-11:
I've been having the same issue with the esp32c6 and I noticed that all of the other examples except from `bas_peripheral_example` had the following:

```rust
/// Size of L2CAP packets
#[cfg(not(feature = "esp"))]
pub const L2CAP_MTU: usize = 128;
#[cfg(feature = "esp")]
// Some esp chips only accept an MTU >= 1017
pub const L2CAP_MTU: usize = 1017;
```

Adding the above to the `bas_peripheral_example` stops it from panicing and I can now see and interact with device via bluetooth.

Note that I'm using a Waveshare ESP32-C6 Development Board that has a ESP32-C6-WROOM-1-N8 Module.

lulf, 2025-01-11:
Yup, this should probably be set for ble_bas_peripheral as well, good catch! Can you create a PR with the fix?

eloycoto, 2025-01-11:
I can confirm that it's working, thanks @sjroe for the hint!

I'll let @sjroe to make the PR for the credit, if he didn't do that in the next couple of days I'll submit it!

@lulf where you can see the bigger values for MTU? I was checking the datasheet yesterday and I didn't spot it.

lulf, 2025-01-11:
Agreed to renamed the feature, i prefer the first variant

eloycoto, 2025-01-13:
Just for context, sometimes when I flash/start again the ble, the init of `esp_wifi::init` got stuck in the interrupts (c6)

https://github.com/esp-rs/esp-hal/blob/7d7ce279d139bfd306f2dee586c198f1821bbf15/esp-wifi/src/timer/timer_esp32c6.rs#L29-L32

Mostly here:

https://github.com/esp-rs/esp-hal/blob/7d7ce279d139bfd306f2dee586c198f1821bbf15/esp-hal/src/interrupt/riscv.rs#L364-L366

When interrupt_number 4, cpu_interrupt_number 1

jamessizeland, 2025-01-14:
Here's my terminal output from running `cargo esp32c6 --bin ble_bas_peripheral` twice on both my waveshare and tinyC6. 

https://gist.github.com/jamessizeland/91cd55d0e80e5995cb413a915ee6849e

jamessizeland, 2025-01-15:
https://github.com/esp-rs/esp-hal/issues/2883 added to esp_wifi issue

eloycoto, 2025-01-16:
Problem has been fixed upstream:

https://github.com/esp-rs/esp-hal/pull/2981

I guess that esp-hal need to be updated to the latest version in examples

eloycoto, 2025-01-17:
closed by c5c2eaba230079662bc9be92e41b279492e78f58, thanks for fixing!

PR #235 (2025-01-12): Increase mtu for ble_bas_peripheral for esp examples
Fixes #229
closed 2025-01-17 by eloycoto