embassy-rs/trouble #234: Q: Expected squashing to be used when PR's are merged

lure23, 2025-01-12:
Just a small question about the management of the git repo.

I recently made some PR's and expected them to be squashed (merged into one change) when being merged with the code base. Then noticed the [original separate commits](https://github.com/embassy-rs/trouble/pull/222/commits) were there:

```
commit e5a6ca8cf850f56d403670ca318d4271aded2bd6
Author: Asko Kauppi <demo@outstanding.earth>
Date:   Wed Jan 8 15:11:56 2025 +0200

    removal of unneeded dep

commit 0e63238b610e3ef1391ac6c7a7da0782f8703a20
Author: Asko Kauppi <demo@outstanding.earth>
Date:   Wed Jan 8 13:34:17 2025 +0200

    'embedded-io-async' dependency seems to be unneeded
```

I'd like to know whether this was by mistake - or what the policy for merging PR for `trouble` is. I don't usually pay so much attention to commit messages when working on something that might get rejected - expecting someone to go through the lines and sum them up at the squash state.

If there is no policy, would it make sense to enforce squashing?

Benefits:
- easier to go back/forth in the git history
- cleaner history since some interim changes disappear

jamessizeland, 2025-01-12:
I agree, I prefer squash merge. I tend to do that, but I don't think it's enforced here yet.

lulf, 2025-01-12:
I tend to prefer rebase merge, assuming each commit is self contained and well written commit message. If not, i will squash merge unless the author does the squashing manually before review or merge.

jamessizeland, 2025-01-12:
Should we turn off standard merge then and just keep squash and rebase options?

lure23, 2025-01-12:
>unless the author does the squashing manually before review or merge.

I'll try to do that in the future, found [this](https://gist.github.com/lpranam/4ae996b0a4bc37448dc80356efbca7fa).

jamessizeland, 2025-01-12:
> >unless the author does the squashing manually before review or merge.
> 
> I'll try to do that in the future, found [this](https://gist.github.com/lpranam/4ae996b0a4bc37448dc80356efbca7fa).

Nice, I'd never considered doing that.

closed 2025-01-17 by lure23