embassy-rs/trouble #245: Proc-macros do not work unless I explicitly import some stuff (as in the trouble prelude)

ivmarkov, 2025-01-17:
Consider:

```rust
// GATT Server definition
#[gatt_server]
struct Server {
    battery_service: BatteryService,
}

/// Battery service
#[gatt_service(uuid = service::BATTERY)]
struct BatteryService {
    /// Battery Level
    #[characteristic(uuid = characteristic::BATTERY_LEVEL, read, notify, value = 10)]
    level: u8,
    #[characteristic(uuid = "408813df-5dd4-1f87-ec11-cdb001100000", write, read, notify)]
    status: bool,
}
```

Once I import the `gatt_server`, `gatt_service`, `characteristic` and so on, rather than the code compiling successfully, it starts to complain that it does not know about:
`AttributeServer`
`AttributeHandle`
`GAP_SERVICE_ATTRIBUTE_COUNT`

... and so on.

I get it, that glob-importing the prelude (`use trouble_host::prelude::*;`) would fix that, but is this idiomatic Rust?
I always thought (a) preludes are optional and (b) code generated with proc-macros should always be using full type names (or do local imports) anyway - if for nothing else - due to potential collisions with the surrounding context where the macros are placed.

jamessizeland, 2025-01-17:
Yeah we're happy to take a look at this to improve the idiomaticness. This is the biggest proc_macro that my team have developed so far, so very happy for any feedback on what can be improved.

@petekubiak

petekubiak, 2025-01-17:
Yes I think this is a very fair point!

closed 2025-01-20 by lulf