embassy-rs/trouble #276: IllegalInstruction running ble_bas_peripheral.rs

Makuo12, 2025-01-28:
When using trouble-host I occasionally get this error

ASSERT_PARAM(0 20), in lld_evt.c at line 2371

Exception occurred 'IllegalInstruction'
Context
PC=0x4008b3fb       PS=0x00060e11
0x4008b3fb - r_assert
    at ??:??
A0=0x8008329e       A1=0x3ffdbb00       A2=0x00000000       A3=0x00000000       A4=0x00000014
A5=0x3f415a2d       A6=0x00000943       A7=0xfffffffb       A8=0x8000814b       A9=0x3ffdba70
A10=0x00000000      A11=0x3ffdba94      A12=0x3ffdba3f      A13=0x00000031      A14=0x00000000
A15=0x3ffdba45
SAR=00000004
EXCCAUSE=0x00000000 EXCVADDR=0x00000000
LBEG=0x400012c5     LEND=0x400012d5     LCOUNT=0xfffffffe
THREADPTR=0x00000000
SCOMPARE1=0x00000100
BR=0x00000000
ACCLO=0x00000000    ACCHI=0x00000000
M0=0x00000000       M1=0x00000000       M2=0x00000000       M3=0x00000000
F64R_LO=0x00000000  F64R_HI=0x00000000  F64S=0x00000000
FCR=0x00000000      FSR=0x00000080
F0=0x433b8000       F1=0x426c0000       F2=0x43800000       F3=0x00000000       F4=0x00000000
F5=0x00000000       F6=0x00000000       F7=0x00000000       F8=0x00000000       F9=0x00000000
F10=0x00000000      F11=0x00000000      F12=0x00000000      F13=0x00000000      F14=0x00000000
F15=0x00000000
0x40084f7e
r_lld_evt_end
    at ??:??
0x40084b71
r_lld_evt_end_isr
    at ??:??
0x4008785d
r_rwble_isr
    at ??:??
0x400882c6
r_rwbtdm_isr_wrapper
    at ??:??
0x400db893
RWBLE
    at /Users/mac/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-wifi-0.12.0/src/timer/timer_esp32.rs:88
0x40080f3c
esp_hal::interrupt::xtensa::vectored::handle_interrupt
    at /Users/mac/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-hal-0.23.1/src/interrupt/xtensa.rs:658
0x40080f01
handle_interrupts
    at ??:??
0x40080872
.RestoreContext
    at ??:??
0x40040001
0x400d930b
core::ptr::read_volatile
    at /Users/mac/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/ptr/mod.rs:1750

lulf, 2025-01-29:
This seems like an esp32-specific issue. Could you provide some more details/logs before the crash?

lure23, 2025-01-29:
@Makuo12 In particular, it would be good to know the type of ESP32 you are using. 

It looks like a thing to report in https://github.com/esp-rs/esp-hal

Makuo12, 2025-01-29:
@lure23 It is public [ble-project](https://github.com/Makuo12/link-x)

lure23, 2025-01-29:
> [@lure23](https://github.com/lure23) It is public [ble-project](https://github.com/Makuo12/link-x)

Had a brief look. 

First, the IllegalInstruction should clearly not occur. But there's more "details" that I think you should provide.

- You haven't mentioned the chip / devkit you use. What does `espflash board-info` tell?
- It's clearly a `nostd` project (using Embassy), but there's a script mentioning `idf`.  How come?

Aside, I'm also working on BLE to mobile, using web browsers (Android Chrome) for UI. But currently have no ESP32 (Xtensa) hardware to actually test yours.

lure23, 2025-01-29:
@Makuo12 Are you on https://matrix.to/#/#esp-rs:matrix.org ?  I would first ask there if `ASSERT_PARAM(0 20)` is familar to anyone. Then open an issue in `esp-hal`. This clearly isn't anything TrouBLE can fix, I think.

closed 2025-01-29 by Makuo12

Makuo12, 2025-01-30:
> Had a brief look.
>
> First, the IllegalInstruction should clearly not occur. But there's more "details" that I think you should provide.
>
> * You haven't mentioned the chip / devkit you use. What does [...]

@lure23 sorry I didn't see the questions. 
I am using the HiLetgo ESP-WROOM-32 Development Board 2.4GHz Dual-Mode WiFi + Bluetooth Dual Cores Microcontroller.
The scripts were automatically generated I am not using them for anything