embassy-rs/trouble #277: Conn.disconnect

Makuo12, 2025-01-28:
I wanted to ask if there is a specific way to use conn.disconnect.
Using the [ble_bas_peripheral.rs](https://github.com/embassy-rs/trouble/blob/main/examples/esp32/src/bin/ble_bas_peripheral.rs) example I created like a timeout async function 

```rust
async fn timeout(conn: &Connection<'_>) {
    loop {
        Timer::after_secs(5).await;
        println!("running timeout");
        conn.disconnect();
    }
}
```
The would run along side gatt_events_task 
```rust
let a = gatt_events_task(
                            &server, &conn, aes, connection_success,
                        read_content, &mut chatacter);
                        let t = timeout(&conn);
let t = timeout(&conn);
match select(a, t).await {}              
```

I realized that when the timeout gets called sometimes the connection disconnects properly. However sometimes it disconnects and stops the ble from advertising again.

lulf, 2025-01-29:
Calling `disconnect()` will not cause it to disconnect immediately, but  cause the ble_task to perform a disconnect on the controller.

Your timeout function will never finish though due to the loop though, it should be sufficient to call disconnect() only once.

Makuo12, 2025-01-29:
So from the logs:
When disconnect gets called from GattEvent::Read
```rust
                    Ok(Some(GattEvent::Read(event))) => {
                        if event.handle() == level.handle {
                            let value = server.get(&level);
                            info!("[gatt] Read Event to Level Characteristic: {:?}", value);
                        } else if event.handle() == status.handle {
                            let value = server.get(&status);
                            info!("[gatt] Read Event to Status Characteristic: {:?}", value);
                            let success;
                            {
                                let result = *connection_success.lock().await;
                                success = result;
                            }
                            if success {
                                info!("has read success");
                                handle_read_state(&mut status, server, read_content, aes).await;
                                {
                                    let mut read = read_content.lock().await;
                                    if read.state == ReadState::Done {
                                        read.state = ReadState::Price;
                                        *character = 'M';
                                        conn.disconnect();
                                        generic_delay(embassy_time::Delay).await;
                                    }
                                    info!("done 2");
                                }
                            } else {
                                info!("has not read success");
                                *character = 'T';
                                conn.disconnect();
                                generic_delay(embassy_time::Delay).await;
                                continue;
                            }
                        }
                    }
```
I notice that INFO - [gatt] disconnected: Err(Connection Terminated By Local Host) gets logged which is correct but after some seconds I get the error log BleHost(NotFound), which is called from ble_task

```rust
async fn ble_task<C: Controller>(mut runner: Runner<'_, C>) -> bool{
    loop {
        if let Err(e) = runner.run().await {
            let e = defmt::Debug2Format(&e);
            panic!("[ble_task] error: {:?}", e);
        }
    }
}
```
This error causes it not to start advertising again

lulf, 2025-01-29:
> So from the logs: I found out that the ble host is no more found. When disconnect gets called from GattEvent::Read
> 
>                     Ok(Some(GattEvent::Read(event))) => {
>                         if event.handle() == level.handle {
>                             let value = server.get(&level);
>                             info!("[gatt] Read Event to Level Characteristic: {:?}", value);
>                         } else if event.handle() == status.handle {
>                             let value = server.get(&status);
>                             info!("[gatt] Read Event to Status Characteristic: {:?}", value);
>                             let success;
>                             {
>                                 let result = *connection_success.lock().await;
>                                 success = result;
>                             }
>                             if success {
>                                 info!("has read success");
>                                 handle_read_state(&mut status, server, read_content, aes).await;
>                                 {
>                                     let mut read = read_content.lock().await;
>                                     if read.state == ReadState::Done {
>                                         read.state = ReadState::Price;
>                                         *character = 'M';
>                                         conn.disconnect();
>                                         generic_delay(embassy_time::Delay).await;
>                                     }
>                                     info!("done 2");
>                                 }
>                             } else {
>                                 info!("has not read success");
>                                 *character = 'T';
>                                 conn.disconnect();
>                                 generic_delay(embassy_time::Delay).await;
>                                 continue;
>                             }
>                         }
>                     }
> 
> I notice that INFO - [gatt] disconnected: Err(Connection Terminated By Local Host) gets logged which is correct but after some seconds I get the error log BleHost(NotFound), which is called from ble_task
> 
> async fn ble_task<C: Controller>(mut runner: Runner<'_, C>) -> bool{
>     loop {
>         if let Err(e) = runner.run().await {
>             let e = defmt::Debug2Format(&e);
>             panic!("[ble_task] error: {:?}", e);
>         }
>     }
> }
> 
> This error causes it not to start advertising again. I know I can just use a warn rather than a panic but I feel the panic is important because it stops you from using the ble when there is no BleHost anymore.

Yes, this is the reason advertising is not running, the runner must be running for it to perform operations to the controller. The host should generally not exit unless there is a condition it cannot recover from. So we should figure out why it is not recovering from this.

Could you share more of the logs? Are any of these two warnings printed? https://github.com/embassy-rs/trouble/blob/main/host/src/host.rs#L923-L929

Also, if you enable tracing, which of these gets printed? https://github.com/embassy-rs/trouble/blob/main/host/src/host.rs#L559-L570

Makuo12, 2025-01-29:
This is the error that gets printed `[host] tx_fut exit`.
```
[ble_task] BleHost(NotFound)

Backtrace:

0x400e0d85
0x400e0d85 - link::run::{{closure}}
    at /Users/mac/Documents/embed_rust/link/src/lib.rs:93
0x400d2369
0x400d2369 - embassy_executor::raw::SyncExecutor::poll::{{closure}}
    at /Users/mac/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-executor-0.7.0/src/raw/mod.rs:430
0x400d43e3
0x400d43e3 - async_main::__xtensa_lx_rt_main
    at /Users/mac/Documents/embed_rust/link/src/bin/async_main.rs:16
0x400efafa
0x400efafa - Reset
    at /Users/mac/.cargo/registry/src/index.crates.io-6f17d22bba15001f/xtensa-lx-rt-0.18.0/src/lib.rs:82
0x400d9067
0x400d9067 - ESP32Reset
    at /Users/mac/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-hal-0.23.1/src/soc/esp32/mod.rs:122

```

lulf, 2025-01-30:
Thanks, I think I know what happens now. Could you try #279 to see if it fixes the issue for you?

Makuo12, 2025-01-30:
Thank you @lulf , I just tried #279 and it works well as expected. It disconnects and continues advertising with no panic errors

lulf, 2025-01-30:
Thanks for testing @Makuo12 !

commit 6a5e1f7f (2025-01-30): fix: do not exit runner on non-critical errors

Connection handles may be marked as disconnected before the runner
has a chance to send responses to it. In that case, log and ignore error.

Fixes #277
PR #279 (2025-01-30): fix: do not exit runner on non-critical errors
Connection handles may be marked as disconnected before the runner has a chance to send responses to it. In that case, log and ignore error.

Fixes #277
commit 77009320 (2025-01-30): fix: do not exit runner on non-critical errors

Connection handles may be marked as disconnected before the runner
has a chance to send responses to it. In that case, log and ignore error.

Fixes #277
closed 2025-01-30 by lulf