embassy-rs/trouble #288: ESP32(C3) scanner example does not work

Meigs2, 2025-02-19:
Hello,

I attempted to port the `ble_scanner.rs` example app over to the ESP32 examples folder by following the previous examples (simply importing and calling `ble_scanner::run::<_>(controller).await;`). It compiles and runs, however nothing appears in the debug output, even though there's plenty of scanable devices nearby, as shown on nRF Connect app on my phone.

Is there something missing for the example to work? Obviously the example is not there in source right now, not sure if there's something I'm missing.

Thanks,
Connor

Meigs2, 2025-02-19:
To save some clicks, I've forked the repo and added the example as I've used it here:

https://github.com/Meigs2/trouble/blob/main/examples/esp32/src/bin/ble_scanner.rs

jamessizeland, 2025-02-19:
Thanks for taking the initiative on this! I've had a go at bringing it up and the solution is the L2CAP_MTU is too low.

If you follow the pattern in the ble_bas_peripheral where L2CAP_MTU is a generic const on run, and pulled through from const.rs in the esp example, it works. It can be set to 27 in the other examples but esp seems to want a larger Maximum Transmission Unit.

Meigs2, 2025-02-20:
Ohhhh, I see. Okay. I'll give that a shot, I saw some discussion around the MTU values recently and I had a suspicion it was related given everything compiles. I'll try that here in a bit and close the issue if this is the case.

Meigs2, 2025-02-20:
Oh yeah, works like a charm. Thanks for the help.