embassy-rs/trouble #295: Not sure how to use trouble-host due to embassy-time

brandonros, 2025-02-22:
I know between all of the components the ecosystem moves very quickly and sometimes a combination of `git` and `crates` needs to be used

```
$ cargo run
    Updating git repository `https://github.com/embassy-rs/trouble`
    Updating crates.io index
error: failed to select a version for `embassy-time-driver`.
    ... required by package `embassy-time v0.4.0`
    ... which satisfies dependency `embassy-time = "^0.4"` of package `trouble-host v0.1.0 (https://github.com/embassy-rs/trouble?rev=b6694cf00b602efe9c1c6e639e97218ebf623479#b6694cf0)`
    ... which satisfies git dependency `trouble-host` of package `rp2350-ble v0.1.0 (/Users/brandon/Desktop/rp2350-ble)`
versions that meet the requirements `^0.2` are: 0.2.0

the package `embassy-time-driver` links to the native library `embassy-time`, but it conflicts with a previous package which links to `embassy-time` as well:
package `embassy-time-driver v0.2.0 (https://github.com/embassy-rs/embassy?rev=2e7a2b6#2e7a2b61)`
    ... which satisfies git dependency `embassy-time-driver` (locked to 0.2.0) of package `embassy-rp v0.3.0 (https://github.com/embassy-rs/embassy?rev=2e7a2b6#2e7a2b61)`
    ... which satisfies git dependency `embassy-rp` (locked to 0.3.0) of package `rp2350-ble v0.1.0 (/Users/brandon/Desktop/rp2350-ble)`
Only one package in the dependency graph may specify the same links value. This helps ensure that only one copy of a native library is linked in the final binary. Try to adjust your dependencies so that only one package uses the `links = "embassy-time"` value. For more information, see https://doc.rust-lang.org/cargo/reference/resolver.html#links.

failed to select a version for `embassy-time-driver` which could resolve this conflict
```

```toml
[dependencies]
embassy-executor = { version = "0.7.0", features = ["task-arena-size-98304", "arch-cortex-m", "executor-thread", "executor-interrupt", "defmt"] }
embassy-futures = { version = "0.1.1" }
embassy-time = { git = "https://github.com/embassy-rs/embassy", rev = "2e7a2b6", features = ["defmt", "defmt-timestamp-uptime"] }
embassy-rp = { git = "https://github.com/embassy-rs/embassy", rev = "2e7a2b6", features = ["defmt", "unstable-pac", "time-driver", "critical-section-impl", "rp235xa", "binary-info"] }
cyw43 = { git = "https://github.com/embassy-rs/embassy", rev = "2e7a2b6", features = ["defmt", "firmware-logs", "bluetooth"] }
cyw43-pio = { git = "https://github.com/embassy-rs/embassy", rev = "2e7a2b6", features = ["defmt"] }
bt-hci = { version = "0.2.1", features = ["embassy-time", "uuid"] }
defmt = "0.3"
panic-probe = { version = "0.3", features = ["print-defmt"] }
defmt-serial = "0.10.0"
cortex-m-rt = "0.7.0"
static_cell = "2.1"
trouble-host = { git = "https://github.com/embassy-rs/trouble", rev = "b6694cf00b602efe9c1c6e639e97218ebf623479", features = ["defmt"] }
```

Could somebody lend a suggestion? I spent a good bit of time trying to get it to work. If I don't use `git` versions of packages, I get no logs/it just hangs.

lure23, 2025-02-22:
I'd take the `rev`s away from `embassy-time` and `embassy-rp`; did you try that?   Not an RP user myself, though.

brandonros, 2025-02-22:
You have to do

```toml
[patch.crates-io]
embassy-time = { git = "https://github.com/embassy-rs/embassy", rev = "2e7a2b6" }
```

to get past this, then you'll get

`Found waker not created by the Embassy executor. `embassy_time::Timer` only works with the Embassy executor.`

which you can do

```toml
[patch.crates-io]
embassy-executor = { git = "https://github.com/embassy-rs/embassy", rev = "2e7a2b6" }
```

I ended up with this in the end, thanks

```toml
[package]
edition = "2021"
name = "rp2350-ble"
version = "0.1.0"
license = "MIT OR Apache-2.0"

[dependencies]
embassy-futures = "0.1.1"
embassy-sync = "0.6.2"
embassy-executor = { version = "*", features = ["task-arena-size-98304", "arch-cortex-m", "executor-thread", "executor-interrupt", "defmt"] }
embassy-time = { version = "*", features = ["defmt", "defmt-timestamp-uptime"] }
embassy-rp = { git = "https://github.com/embassy-rs/embassy", rev = "2e7a2b6", features = ["defmt", "unstable-pac", "time-driver", "critical-section-impl", "rp235xa", "binary-info"] }
cyw43 = { git = "https://github.com/embassy-rs/embassy", rev = "2e7a2b6", features = ["defmt", "firmware-logs", "bluetooth"] }
cyw43-pio = { git = "https://github.com/embassy-rs/embassy", rev = "2e7a2b6", features = ["defmt"] }
bt-hci = { version = "*", features = ["embassy-time", "uuid", "defmt"] }
trouble-host = { git = "https://github.com/embassy-rs/trouble", rev = "b6694cf", features = ["defmt"] }
defmt = "0.3"
panic-probe = { version = "0.3", features = ["print-defmt"] }
defmt-serial = "0.10.0"
cortex-m-rt = "0.7.0"
static_cell = "2.1"

[patch.crates-io]
embassy-time = { git = "https://github.com/embassy-rs/embassy", rev = "2e7a2b6" }
embassy-executor = { git = "https://github.com/embassy-rs/embassy", rev = "2e7a2b6" }
bt-hci = { git = "https://github.com/embassy-rs/bt-hci", rev = "21ba1ce" }

[features]
default = ["defmt"]
defmt = ["embassy-time/defmt", "embassy-rp/defmt", "cyw43/defmt", "bt-hci/defmt"]

[profile.release]
debug = 2

[profile.dev]
lto = true
opt-level = "z"
```

closed 2025-02-22 by brandonros