embassy-rs/trouble #296: ESP32x alloc panic using central/runner after bluetooth scan?

Meigs2, 2025-02-23:
Hello,

I've been attempting to modify the ble_scanner example to try and get the scanner to connect to a bluetooth scale of mine, which I don't know the address of, by name. It's going well, however, for some reason I'm getting an allocation panic after I find the correct device by name.

Here is my modified example code for `ble_scanner.rs`, invoked using `cargo espflash flash --monitor --target riscv32imc-unknown-none-elf --bin ble_scanner --features esp32c3`:

```rust
use bt_hci::cmd::le::LeSetScanParams;
use bt_hci::controller::ControllerCmdSync;
use core::cell::RefCell;
use embassy_futures::join::join;
use embassy_futures::select::select;
use embassy_sync::blocking_mutex::raw::CriticalSectionRawMutex;
use embassy_sync::signal::Signal;
use embassy_time::{Duration, Timer};
use heapless::{Deque, String, Vec};
use trouble_host::prelude::*;

/// Max number of connections
const CONNECTIONS_MAX: usize = 1;
const L2CAP_CHANNELS_MAX: usize = 1;

pub async fn run<C, const L2CAP_MTU: usize>(controller: C)
where
    C: Controller + ControllerCmdSync<LeSetScanParams>,
{
    // Using a fixed "random" address can be useful for testing. In real scenarios, one would
    // use e.g. the MAC 6 byte array as the address (how to get that varies by the platform).
    let address: Address = Address::random([0xff, 0x8f, 0x1b, 0x05, 0xe4, 0xff]);

    info!("Our address = {:?}", address);
    let mut resources: HostResources<CONNECTIONS_MAX, L2CAP_CHANNELS_MAX, L2CAP_MTU> = HostResources::new();
    let stack = trouble_host::new(controller, &mut resources).set_random_address(address);
    let Host {
        mut central, mut runner, ..
    } = stack.build();

    let (central, runner, addr, addr_kind) = {
        let printer = Printer {
            seen: RefCell::new(Deque::new()),
            found: Signal::new(),
        };

        let mut scanner = Scanner::new(central);
        let _ = select(runner.run_with_handler(&printer), async {
            let mut config = ScanConfig::default();
            config.active = true;
            config.phys = PhySet::M1;
            config.interval = Duration::from_secs(1);
            config.window = Duration::from_secs(1);
            let mut _session = scanner.scan(&config).await.unwrap();
            // Scan forever
            loop {
                if !printer.found.signaled() {
                    Timer::after(Duration::from_secs(1)).await;
                    continue;
                }

                break;
            }
        })
            .await;

        let Some(value) = printer.found.try_take() else {
            panic!("Idk how you got here but you did. Good job.")
        };

        let central = scanner.into_inner();
        let runner = runner;

        (central, runner, value.0, value.1)
    };

    info!("Found the scale!");

    loop {
        info!("Spin looping.");
        Timer::after(Duration::from_secs(1)).await; // Panic here after ~3 spin-loops?
    }
}

struct Printer {
    seen: RefCell<Deque<BdAddr, 128>>,
    found: Signal<CriticalSectionRawMutex, (BdAddr, AddrKind)>,
}

impl EventHandler for Printer {
    fn on_adv_reports(&self, mut it: LeAdvReportsIter<'_>) {
        let mut seen = self.seen.borrow_mut();
        while let Some(Ok(report)) = it.next() {
            if seen.iter().find(|b| b.raw() == report.addr.raw()).is_none() {
                let mut advertisements = AdStructure::decode(report.data);
                while let Some(Ok(ad)) = advertisements.next() {
                    match ad {
                        AdStructure::CompleteLocalName(n) => {
                            let value = get_name_string(n);
                            info!("Short Name: {:?}", value);
                            if value.starts_with("ACAIA") {
                                info!("SIGNALING");
                                self.found.signal((report.addr, report.addr_kind))
                            }
                        }
                        AdStructure::ShortenedLocalName(n) => {
                            let value = get_name_string(n);
                            info!("Short Name: {:?}", value);
                            if value.starts_with("ACAIA") {
                                info!("SIGNALING");
                                self.found.signal((report.addr, report.addr_kind))
                            }
                        }
                        _ => {}
                    }
                }

                if seen.is_full() {
                    seen.pop_front();
                }
                seen.push_back(report.addr).unwrap();
            }
        }
    }
}

fn get_name_string(slice: &[u8]) -> String::<291> {
    let mut vec = Vec::<u8, 291>::new();
    vec.extend_from_slice(&slice).unwrap();
    String::from_utf8(vec).unwrap()
}
```

Key changes are in `Printer`, extracting out the `AdStructure` to check the name string, and swapping out `join` for `select` to exit the scanning loop. I then re-extract the `central` and  'runner` peripherals to be used again to connect/do whatever.

This all seemingly works fine, and I just put a spin-loop after to make sure everything works, but after it starts spin-looping, I get an allocator panic:

```
INFO - Short Name: "ACAIAL-C1F45CE"
INFO - SIGNALING
INFO - Found the scale!
INFO - Spin looping.
INFO - Spin looping.
INFO - Spin looping.

====================== PANIC ======================
panicked at library/alloc/src/alloc.rs:438:13:
memory allocation of 8192 bytes failed

Backtrace:

0x4201513e
0x4201513e - alloc::alloc::handle_alloc_error::rt_error
    at /rustc/4d91de4e48198da2e33413efdcd9cd2cc0c46688/library/alloc/src/alloc.rs:403
0x42015128
0x42015128 - alloc::raw_vec::handle_error
    at /rustc/4d91de4e48198da2e33413efdcd9cd2cc0c46688/library/alloc/src/raw_vec.rs:792
0x42011fa0
0x42011fa0 - alloc::raw_vec::RawVecInner<A>::grow_one
    at C:\Users\Connor\.rustup\toolchains\stable-x86_64-pc-windows-msvc\lib/rustlib/src/rust\library/alloc/src\raw_vec.rs:566
0x42011d4c
0x42011d4c - alloc::collections::vec_deque::VecDeque<T,A>::grow
    at C:\Users\Connor\.rustup\toolchains\stable-x86_64-pc-windows-msvc\lib/rustlib/src/rust\library/alloc/src/collections/vec_deque\mod.rs:2192
0x42011e98
0x42011e98 - alloc::collections::vec_deque::VecDeque<T,A>::push_back
    at C:\Users\Connor\.rustup\toolchains\stable-x86_64-pc-windows-msvc\lib/rustlib/src/rust\library/alloc/src/collections/vec_deque\mod.rs:1782
0x42012244
0x42012244 - esp_wifi::ble::btdm::notify_host_recv::{{closure}}
    at C:\Users\Connor\.cargo\git\checkouts\esp-hal-29ef8683203c9b81\5d0145e\esp-wifi\src\ble\btdm.rs:79
```

I don't quite understand what to do here, or if this is something I even need to worry about. Seems related to logging? The target this seems to be panicing on is for `stable-x86_64-pc-windows-msvc`, which is obviously not `riscv32imc-unknown-none-elf`, but I assume it's espflash propagating out here. 

If I put the entire `resources` declaration in a scope and then spinloop after after the `Stack`/`Host` are dropped, everything is fine. So it seems like the central/runner are still doing stuff in the background even after the future is dropped, which I would not expect as behavior. If this is expected, or if I need to do something different to get something like this working, please let me know.

Thanks.

lulf, 2025-02-23:
I believe this is related to esp32 hal controller implementation. You could try to increase the memory available on the esp32 heap by adjusting this value https://github.com/embassy-rs/trouble/blob/main/examples/esp32/src/bin/ble_bas_peripheral.rs#L22

I'm not sure if that will just postpone the issue or if you need to somehow stop scanning to avoid the controller being overfloded by scan reports or something.

Meigs2, 2025-02-23:
> I believe this is related to esp32 hal controller implementation. You could try to increase the memory available on the esp32 heap by adjusting this value [...]

I guess this is more of a behavioral question, but I'd expect that if the future created by `runner.run_with_handler()` is dropped, that the controller would stop scanning. I'm not sure how that's handled on a technical level with the ESP32 + trouble. I do live in an apartment and there's a ton of devices around.

It should also be noted that there's no issues with scanning and just leaving it loop, this happens specifically when dropping that scan future. I can sit and scan devices all day, unless I flood the debug output and the allocator can't keep up.

lulf, 2025-02-24:
In order to stop scanning, you need to drop the session handle here:

```
            let mut _session = scanner.scan(&config).await.unwrap();
```

The runner will then process the request to drop the scanning, which will cause the controller to stop spamming that. 

In general the runner should not be stopped until you're sure there are no BLE activity going on.