embassy-rs/trouble #303: Moving to Rust 2024 breaks ESP32 builds

sssilver, 2025-02-27:
Looks like TrouBLE has moved to the 2024 edition.

Unfortunately the ESP32 toolchain doesn't seem to support it yet, which means we need to change our `rust-toolchain.toml` from:
```
[toolchain]
channel = "esp"
```
to:
```
[toolchain]
channel = "1.85"
```

Completely breaking the build:

```
...
'esp32-s3' is not a recognized processor for this target (ignoring processor)
'esp32-s3' is not a recognized processor for this target (ignoring processor)
'esp32-s3' is not a recognized processor for this target (ignoring processor)
'esp32-s3' is not a recognized processor for this target (ignoring processor)
error[E0463]: can't find crate for `core`
  |
  = note: the `xtensa-esp32s3-espidf` target may not be installed
  = help: consider downloading the target with `rustup target add xtensa-esp32s3-espidf`

For more information about this error, try `rustc --explain E0463`.
error: could not compile `bitflags` (lib) due to 1 previous error
warning: build failed, waiting for other jobs to finish...

  $ rustup target add xtensa-esp32s3-espidf
error: toolchain '1.85-aarch64-apple-darwin' does not support target 'xtensa-esp32s3-espidf'
note: you can see a list of supported targets with `rustc --print=target-list`
note: if you are adding support for a new target to rustc itself, see https://rustc-dev-guide.rust-lang.org/building/new-target.html
```

What's the guidance in this situation?

jamessizeland, 2025-02-27:
Apologies! We should have checked for that interaction. It didn't occur to me that it would do that.

I think if you can, pin your commit of trouble to rev="b6694cf00b602efe9c1c6e639e97218ebf623479"

Which was the last commit before 2024, and we'll get an idea from the esp team on when the Xtensa fork will support 2024.

I'll try this on my S3 when I get a chance.

lulf, 2025-02-27:
Yeah, sorry for that. We should add xtensa to CI if we can.

sssilver, 2025-02-27:
Should we also introduce semantic versioning to help identify expected breaking changes, and start publishing to crates.io?

lulf, 2025-02-27:
Thats the plan. Just waiting for security manager to land and writing a little more docs first.

commit 5286a639 (2025-03-19): fix: move back to 2021 edition

Downgrade for now until xtensa supports 2021.

Fixes #303
commit 84051feb (2025-03-19): feat: enable phy update event and log error (#318)

* feat: enable phy update event and log error

* fix: move back to 2021 edition

Downgrade for now until xtensa supports 2021.

Fixes #303

* chore: rustfmt