embassy-rs/trouble #305: BleHost(OutOfMemory)

Easyoakland, 2025-03-02:
I'm trying to subscribe to a heart-rate monitor's characteristic.

I'm using an esp32 on the 54cd0465ca497495b5890d896fc1f048f0a0f1ee revision. It doesn't look like there's an update related to my issue, so it shouldn't matter.

The relevant (simplified) code is:
```rs
let connection = match central.connect(config).await.unwrap();
let client = GattClient::<_, 10, { consts::BLE_L2CAP_MTU }>::new(stack, &connection).await.unwrap();
    let ((), ()) = join(async { client.task().await.unwrap() }, async {
        let services = client.services_by_uuid(&consts::BLE_SERIVCE_UUID_16).await.unwrap();
        assert_eq!(services.len(), 1, "todo: != 1");
        let service = services.first().unwrap();

        let characteristic: Characteristic<heapless::Vec<u8, 8>> = client
            .characteristic_by_uuid(service, &consts::BLE_CHARACTERISTIC_UUID_LONG)
            .await
            .unwrap(); // <-------
        let mut listener = client
            .subscribe(&characteristic, false)
            .await
            .unwrap();
    })
    .await;
```

I'm getting an error at the above where `<------` is indicated with `BleHost(OutOfMemory)`.

It looks like this originates here
https://github.com/embassy-rs/trouble/blob/92841fcc8bb986368456239a7bbf25be2f709ec1/host/src/gatt.rs#L622

I'm not sure what this code is doing or what a cccd is, but it seems strange that converting a byte slice to a u16 fails with OutOfMemory. There's not enough information here to determine what is actually broken, which makes the error unhelpful.

If the issue is that a buffer needs to be changed, what is that buffer and how is it increased?

If it helps, I've added a log with debug prints below.

<details><summary>log with added prints</summary>

```
INFO - services [ServiceHandle { start: 40, end: 65535, uuid: Uuid16([13, 24]) }]
INFO - [C:\path\trouble\host\src\gatt.rs:531] pdu request start
INFO - [C:\path\trouble\host\src\gatt.rs:783] att: Server(Response(ReadByType { it: ReadByTypeIter { item_len: 21, cursor: ReadCursor { pos: 2, data: [9, 21, 41, 0, 16, 42, 0, 251, 52, 155, 95, 128, 0, 0, 128, 0, 16, 0, 0, 55, 42, 0, 0] } } }))
INFO - [C:\path\trouble\host\src\gatt.rs:536] match start on ReadByType { it: ReadByTypeIter { item_len: 15, cursor: ReadCursor { pos: 2, data: [9, 15, 29, 0, 10, 2a, 0, fb, 34, 9b, 5f, 80, 0, 0, 80, 0, 10, 0, 0, 37, 2a, 0, 0] } } }
INFO - [C:\path\trouble\host\src\att.rs:400] cursor: ReadCursor { pos: 2, data: [9, 21, 41, 0, 16, 42, 0, 251, 52, 155, 95, 128, 0, 0, 128, 0, 16, 0, 0, 55, 42, 0, 0] }
INFO - [C:\path\trouble\host\src\att.rs:403] handle,item=(29, [10, 2a, 0, fb, 34, 9b, 5f, 80, 0, 0, 80, 0, 10, 0, 0, 37, 2a, 0, 0])
INFO - [C:\path\trouble\host\src\gatt.rs:544] item.len() >=5, decoding declaration
INFO - [C:\path\trouble\host\src\gatt.rs:551] decoded declaration
INFO - [C:\path\trouble\host\src\gatt.rs:557] calling get_characteristic_cccd
INFO - [C:\path\trouble\host\src\gatt.rs:783] att: Server(Response(ReadByType { it: ReadByTypeIter { item_len: 2, cursor: ReadCursor { pos: 2, data: [9, 2, 43, 0] } } }))
INFO - [C:\path\trouble\host\src\att.rs:400] cursor: ReadCursor { pos: 2, data: [9, 2, 43, 0] }
INFO - [C:\path\trouble\host\src\att.rs:403] handle,item=(2b, [])
INFO - [C:\path\trouble\host\src\gatt.rs:601] item: []
[31m

====================== PANIC ======================
```
</details>

Easyoakland, 2025-03-03:
Are the prints in the spoiler tag not what you mean?

lulf, 2025-03-03:
Apologies, I missed the debug print. I'm not sure if this is an issue with the trouble or the device you're querying not adhering to the specification (i.e. can a CCCD value be empty? or is this an issue in the decoding in trouble). Maybe trouble should just set flags to '0' if the value is missing.

lulf, 2025-03-03:
It's been a while since I looked at this code, and I'm not very familiar with the attribute protocol. Each characteristic has a declaration which describes a set of properties https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Core-54/out/en/host/generic-attribute-profile--gatt-.html#UUID-cc38bdec-c550-d1a5-dacd-b9c44345d08d

According to the table in 3.3.1.1, in the table for Notify or Indicate, it says "If set, the Client Characteristic Configuration Descriptor shall exist."

So the code finds those properties, and then proceeds to read the CCCD characteristic. The CCCD is defined in 3.3.3.3 (https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Core-54/out/en/host/generic-attribute-profile--gatt-.html#UUID-09487be3-178b-eeca-f49f-f783e8d462f6).

According to the specification:

"
The characteristic descriptor is contained in an Attribute. The Attribute Type shall be set to the UUID for «Client Characteristic Configuration». The Attribute Value shall be two octets in length and shall be set to the characteristic descriptor value. The Attribute Permissions are specified by the profile or may be implementation specific if not specified otherwise.
"

So it seems like an empty value here is not correct.

The next thing I think is to decode the actual response payload you get there, and verify that the trouble parsing of that payload is correct. 

If you add a print of the raw bytes of the response (in the pdu field) it should be possible to reproduce and examine further.

I agree that a more appropriate error would perhaps be InvalidSize or something like that.

Easyoakland, 2025-03-08:
Looks like that was the issue. Using a default of `0` on an empty read resolves the issue. I'll make a pr shortly.
Here's a screenshot of wireshark on the relevant packet

[image]

closed 2025-03-08 by Easyoakland