embassy-rs/trouble #306: Allow splitting L2capChannel into a read half and a write half.

Dominaezzz, 2025-03-02:
I'm implementing IPSP/[RFC7668](https://datatracker.ietf.org/doc/html/rfc7668) on top of trouble.

To do so I need to be able to read and write data (IPv6 packets) concurrently.

I'm currently doing something like this:
```Rust
let result = select(
    // Wait for network stack/embassy-net to transmit a packet.
    tx.tx_buf(),
    // Wait for packet to come in the L2CAP channel.
    self.channel.receive(&self.stack, &mut buffer),
).await;
```

Whilst it works for a POC, the problem with it is `channel.receive` isn't cancellation friendly.
If `tx.tx_buf()` returns while `self.channel.receive` is reassembling PDUs, this received data will be lost when it's cancelled.

I don't specifically need to split the channel, I just need some way to read and write concurrently.

lulf, 2025-03-03:
You can already .clone() an l2cap channel, is that not what you're looking for? Although now that you've mentioned it, I think I'd like to change the api to use a split() in the future.

Dominaezzz, 2025-03-03:
> You can already .clone() an l2cap channel, is that not what you're looking for?

Whoops, didn't notice that. Yeah that would work.

I'm curious what behaviour to expect if two clones read (or write) at the same time. Will I get garbage or will each come randomly get a full packet?

Splitting would make this more intuitive I think

closed 2025-03-03 by lulf