embassy-rs/trouble #320: Use value directly when expanding characteristic macro

HaoboGu, 2025-03-20:
I'm trying to create a HID service using trouble-host-macro:

```rust
const repost_desc: [u8; 90] = [...];
#[gatt_service(uuid = service::HUMAN_INTERFACE_DEVICE)]
struct HidService {
    #[characteristic(uuid = "2a4a", read, value = [0x01, 0x01, 0x00, 0x03])]
    hid_info: [u8; 4],
    #[characteristic(uuid = "2a4b", read, value = report_desc)]
    report_map: [u8; 90],
    #[characteristic(uuid = "2a4c", write_without_response)]
    hid_control_point: u8,
    #[characteristic(uuid = "2a4e", read, write_without_response, value = 1)]
    protocol_mode: u8,
    #[descriptor(uuid = "2908", read, value = [BleCompositeReportType::Keyboard as u8, 1u8])]
    #[characteristic(uuid = "2a4d", read, notify)]
    input_keyboard: [u8; 8],
    #[descriptor(uuid = "2908", read, value = [BleCompositeReportType::Keyboard as u8, 2u8])]
    #[characteristic(uuid = "2a4d", read, write, write_without_response)]
    output_keyboard: [u8; 1],
}
```
But I got the following error:

```
error[E0599]: no function or associated item named `default` found for array `[u8; 90]` in the current scope
  --> /Users/haobogu/Projects/keyboard/rmk/rmk/src/ble/trouble.rs:84:17
   |
84 |     report_map: [u8; 90],
```

I had to wrap the slice with a local struct and implement default trait for it. 

The macro expansion code is here:

https://github.com/embassy-rs/trouble/blob/84051feb1c43b5b8702a120107e715916d0422ab/host-macros/src/service.rs#L198-L207

I found it's unnecessary to create a default value then re-binding the user provided value, is there any other consideration to not use something like:

```rust
        let default_value = match characteristic.args.default_value {
            Some(val) => quote!(#val),        // if set by user
            None => quote!(<#ty>::default()), // or default otherwise
        };

        self.code_build_chars.extend(quote_spanned! {characteristic.span=>
            let (#char_name, #(#named_descriptors),*) = {
                static #name_screaming: static_cell::StaticCell<[u8; <#ty as trouble_host::types::gatt_traits::AsGatt>::MAX_SIZE]> = static_cell::StaticCell::new();
                let mut val = #default_value  // <- Use default value above directly
```

BTW, is it possible to use a `&static [u8]` as the value here?

lulf, 2025-03-20:
Ah, that's a good fine. The reason it fails is this https://doc.rust-lang.org/std/primitive.array.html

`Arrays of sizes from 0 to 32 (inclusive) implement the [Default](https://doc.rust-lang.org/std/default/trait.Default.html) trait if the element type allows it. As a stopgap, trait implementations are statically generated up to size 32.`

So I think macros need to do something else than rely on default() here.

lulf, 2025-03-20:
As a workaround, perhaps you can use a `Vec<u8, 90>`

PR #322 (2025-03-22): Use user provided value directly when expanding characteristic macro
Closes #320
closed 2025-03-22 by petekubiak