embassy-rs/trouble #338: `BleHost(Disconnected)` when setting connection param

HaoboGu, 2025-04-02:
I'm testing nrf-sdc updating connection parameter after GattConnection is established. I use
```rust
conn.raw().update_connection_params(&stack,
                &ConnectParams {
                    min_connection_interval: Duration::from_micros(7500),
                    max_connection_interval: Duration::from_micros(7500),
                    max_latency: 99,
                    event_length: Duration::from_secs(0),
                    supervision_timeout: Duration::from_secs(5),
                })` 
```
to update connections parameters. But this function always returns `BleHost(Disconnected)`. I'm not sure that whether the connection parameters are updated, but according to 

https://github.com/embassy-rs/trouble/blob/d17132618413f0fa840fee32af4ab23ca125bec2/host/src/connection.rs#L415-L417

it's because that the conn handler is not recognized by the controller, which seems wrong. The data exchange of the connection runs well before and after calling `update_connection_params`

lulf, 2025-04-02:
Is this on the central side or on the peripheral side you're calling this? I'm using this successfully, so I'm surprised that you get this error. Could you share more on how you create the GattConnection?

HaoboGu, 2025-04-02:
On peripheral side. 

I create the connection use:

```rust
async fn advertise<'a, 'b, C: Controller>(
    name: &'a str,
    peripheral: &mut Peripheral<'a, C>,
    server: &'b Server<'_>,
) -> Result<GattConnection<'a, 'b>, BleHostError<C::Error>> {
    // Wait for 10ms to ensure the USB is checked
    embassy_time::Timer::after_millis(10).await;
    let mut advertiser_data = [0; 31];
    AdStructure::encode_slice(
        &[
            AdStructure::Flags(LE_GENERAL_DISCOVERABLE | BR_EDR_NOT_SUPPORTED),
            AdStructure::ServiceUuids16(&[BATTERY.to_le_bytes(), HUMAN_INTERFACE_DEVICE.to_le_bytes()]),
            AdStructure::CompleteLocalName(name.as_bytes()),
            AdStructure::Unknown {
                ty: 0x19, // Appearance
                data: &KEYBOARD.to_le_bytes(),
            },
        ],
        &mut advertiser_data[..],
    )?;

    let advertise_config = AdvertisementParameters {
        primary_phy: PhyKind::Le2M,
        secondary_phy: PhyKind::Le2M,
        tx_power: TxPower::Plus8dBm,
        timeout: Some(Duration::from_secs(120)),
        interval_min: Duration::from_millis(200),
        interval_max: Duration::from_millis(200),
        ..Default::default()
    };

    info!("[adv] advertising");
    let advertiser = peripheral
        .advertise(
            &advertise_config,
            Advertisement::ConnectableScannableUndirected {
                adv_data: &advertiser_data[..],
                scan_data: &[],
            },
        )
        .await?;
    let conn = advertiser.accept().await?.with_attribute_server(server)?;
    info!("[adv] connection established");
    Ok(conn)
}
```

The attribute server is a little bit complex, can be found here:

https://github.com/HaoboGu/rmk/blob/98af3181d2d06ecd6335840319e4a585a1ec25eb/rmk/src/ble/trouble/ble_server.rs#L10

HaoboGu, 2025-04-02:
I also got some attribute not found error, when connecting to linux:

[image]

The linux machine reports `malformed att response`

[image]

HaoboGu, 2025-04-02:
[tmp.log](https://github.com/user-attachments/files/19568896/tmp.log)

The full log which can be opened by wireshark

lulf, 2025-04-02:
Forgot to ask, which controller, nrf-sdc?

HaoboGu, 2025-04-02:
> Forgot to ask, which controller, nrf-sdc?

yes, nrf-sdc

HaoboGu, 2025-04-03:
After more investigation, I found this config only fails on Linux and Android. It works on macOS, iOS and Windows.

lulf, 2025-04-03:
Hi, you might want to check the nrf softdevice controller docs for this command: https://github.com/nrfconnect/sdk-nrfxlib/blob/main/softdevice_controller/include/sdc_hci_cmd_le.h#L3471-L3477

```
  If the Host issues this command on a Connection_Handle where the Controller is the
  Peripheral and either the local or peer Controller does not support the Connection
  Parameters Request procedure (see [Vol 6] Part B, Section 5.1.7), then the Controller
  shall return an error which should use the error code Unsupported Feature or
  Parameter Value (0x11) if the local Controller does not support that procedure and
  Unsupported Remote Feature (0x1A) if the local Controller supports that procedure but
  the peer Controller does not.
 ```

So could be that there are some bluetooth version differences.

HaoboGu, 2025-04-04:
It's quite weird, sometimes I get `UnknownHCICommand` as well:

```
ERROR [set_conn_params] error: "BleHost(Hci(Unknown HCI Command))"
└─ rmk::ble::trouble::set_conn_params::{async_fn#0} @ /Users/haobogu/Projects/keyboard/rmk/rmk/src/ble/trouble/mod.rs:483 
```

It seems that different centrals(windows/linux/..) will result in different error message. I totally have no idea about what happened..

HaoboGu, 2025-04-22:
Just did some more testing. On macOS/iOS, it returns `BleHost(Disconnected)`, and on Linux/Android it returns `BleHost(UnknownHCICommand)`.

lulf, 2025-06-17:
Interesting find! You can use https://github.com/embassy-rs/trouble/blob/main/host/src/host.rs#L498 to send an l2cap signal packet (probably makes sense to do that inside the connection.rs or host.rs in some utility fn that checks the supported features). To encode the signal packet, you define a type and implement the trait, see here: https://github.com/embassy-rs/trouble/blob/main/host/src/types/l2cap.rs#L126

The logic should probably use these commands to check what features are supported by the connection first:

https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Core-54/out/en/host-controller-interface/host-controller-interface-functional-specification.html#UUID-3ad76ac5-3812-cca4-a4f5-f73e96cebcba

https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Core-54/out/en/host-controller-interface/host-controller-interface-functional-specification.html#UUID-86223376-28a9-e454-15f2-e420aee8c462