embassy-rs/trouble #342: Got this PANIC error when running "ble_bas_peripheral" on ESP32 Wrover Kit.

artisdom, 2025-04-07:
```
panicked at C:\Users\Ted\.cargo\git\checkouts\esp-hal-29ef8683203c9b81\5d0145e\esp-println\src\logger.rs:22:42:
called `Result::unwrap()` on an `Err` value: SetLoggerError(())
```

Full log below, any idea, thanks. :)

```
PS C:\r\trouble\examples\esp32
> cargo +esp run --release --no-default-features --features=esp32 --target=xtensa-esp32-none-elf --bin ble_bas_peripheral
    Finished `release` profile [optimized + debuginfo] target(s) in 0.55s
     Running `espflash flash --monitor target\xtensa-esp32-none-elf\release\ble_bas_peripheral`
[2025-04-07T01:03:05Z INFO ] Detected 4 serial ports
[2025-04-07T01:03:05Z INFO ] Ports which match a known common dev board are highlighted
[2025-04-07T01:03:05Z INFO ] Please select a port
[2025-04-07T01:03:13Z INFO ] Serial port: 'COM6'
[2025-04-07T01:03:13Z INFO ] Connecting...
[2025-04-07T01:03:14Z INFO ] Using flash stub
Chip type:         esp32 (revision v3.0)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC address:       34:94:54:0d:4e:78
App/part. size:    380,960/4,128,768 bytes, 9.23%
[2025-04-07T01:03:16Z INFO ] Segment at address '0x1000' has not changed, skipping write
[2025-04-07T01:03:16Z INFO ] Segment at address '0x8000' has not changed, skipping write
[2025-04-07T01:03:16Z INFO ] Segment at address '0x10000' has not changed, skipping write
[2025-04-07T01:03:16Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7104
load:0x40078000,len:15576
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3876
entry 0x4008064c
0x4008064c - __user_exception
    at C:\Users\Ted\.cargo\git\checkouts\esp-hal-29ef8683203c9b81\5d0145e\esp-backtrace\src\lib.rs:103
I (31) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (31) boot: compile time Jun  7 2023 07:48:23
I (33) boot: Multicore bootloader
I (37) boot: chip revision: v3.0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 003f0000
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3ffc0000 size=00314h (   788) load
I (107) esp_image: segment 1: paddr=0001033c vaddr=3ffdb330 size=002ech (   748) load
I (115) esp_image: segment 2: paddr=00010630 vaddr=40080000 size=08ec4h ( 36548) load
I (139) esp_image: segment 3: paddr=000194fc vaddr=3f4094fc size=0b088h ( 45192) map
I (155) esp_image: segment 4: paddr=0002458c vaddr=40088ec4 size=021b8h (  8632) load
I (159) esp_image: segment 5: paddr=0002674c vaddr=00000000 size=098cch ( 39116)
I (175) esp_image: segment 6: paddr=00030020 vaddr=400d0020 size=3cfdch (249820) map
I (270) boot: Loaded app from partition at offset 0x10000
I (271) boot: Disabling RNG early entropy source...

[... 126 lines cut ...]
====================== PANIC ======================
panicked at C:\Users\Ted\.cargo\git\checkouts\esp-hal-29ef8683203c9b81\5d0145e\esp-println\src\logger.rs:22:42:
called `Result::unwrap()` on an `Err` value: SetLoggerError(())

Backtrace:

0x400e45e9
0x400e45e9 - core::result::unwrap_failed
    at C:\Users\Ted\.rustup\toolchains\esp\lib\rustlib\src\rust\library\core\src\result.rs:1704
0x400d96cf
0x400d96cf - core::result::Result<T,E>::unwrap
    at C:\Users\Ted\.rustup\toolchains\esp\lib\rustlib\src\rust\library\core\src\result.rs:1109
0x400d7e69
0x400d7e69 - embassy_executor::raw::SyncExecutor::poll::{{closure}}
    at C:\Users\Ted\.cargo\registry\src\index.crates.io-1949cf8c6b5b557f\embassy-executor-0.7.0\src\raw\mod.rs:430
0x400e2bec
0x400e2bec - ble_bas_peripheral::__xtensa_lx_rt_main
    at C:\r\trouble\examples\esp32\src\bin\ble_bas_peripheral.rs:14
0x400f6a80
0x400f6a80 - Reset
    at C:\Users\Ted\.cargo\git\checkouts\esp-hal-29ef8683203c9b81\5d0145e\xtensa-lx-rt\src\lib.rs:82
0x400e8ce9
0x400e8ce9 - ESP32Reset
    at C:\Users\Ted\.cargo\git\checkouts\esp-hal-29ef8683203c9b81\5d0145e\esp-hal\src\soc\esp32\mod.rs:122
```

artisdom, 2025-04-07:
btw, also tried the same example on my ESP32C6, it seems running fine.

```
cargo run --release --no-default-features --features=esp32c6 --target=riscv32imac-unknown-none-elf --bin ble_bas_peripheral
```

lulf, 2025-04-10:
The error seems unrelated to trouble to me. Have you been able to run it with the examples from esp-hal? Could be there are some features or parameters that need changing.

bjoernQ, 2025-04-14:
I tried the example on an ESP32 DevKitC and it works fine for me
```
I (31) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (31) boot: compile time Jun  7 2023 07:48:23
I (33) boot: Multicore bootloader
I (37) boot: chip revision: v3.1
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 003f0000
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0a84ch ( 43084) map
I (122) esp_image: segment 1: paddr=0001a874 vaddr=3ffc0000 size=00ce8h (  3304) load
I (124) esp_image: segment 2: paddr=0001b564 vaddr=3ffc9cec size=002ech (   748) load
I (128) esp_image: segment 3: paddr=0001b858 vaddr=40080000 size=047c0h ( 18368) load
I (144) esp_image: segment 4: paddr=00020020 vaddr=400d0020 size=3dd94h (253332) map
I (236) esp_image: segment 5: paddr=0005ddbc vaddr=400847c0 size=0657ch ( 25980) load
I (253) boot: Loaded app from partition at offset 0x10000
I (253) boot: Disabling RNG early entropy source...
INFO - Our address = Address { kind: AddrKind(1), addr: BdAddr([255, 143, 26, 5, 228, 255]) }
INFO - Starting advertising and GATT service
INFO - [host] filter accept list size: 12
INFO - [host] setting txq to 10, fragmenting at 251
INFO - [host] configuring host buffers (8 packets of size 1512)
INFO - [host] initialized
INFO - [host] Device Address 8A:13:BF:04:95:B1
INFO - [adv] advertising
INFO - [adv] connection established
INFO - [custom_task] notifying connection of tick 1
INFO - [custom_task] RSSI: -47
INFO - [custom_task] notifying connection of tick 2
INFO - [custom_task] RSSI: -45
INFO - [custom_task] notifying connection of tick 3
INFO - [custom_task] RSSI: -45
INFO - [gatt] Read Event to Level Characteristic: Ok(3)
INFO - [custom_task] notifying connection of tick 4
INFO - [custom_task] RSSI: -48
INFO - [custom_task] notifying connection of tick 5
INFO - [custom_task] RSSI: -43
INFO - [host] disconnection event on handle 0, reason: Remote User Terminated Connection
INFO - [gatt] disconnected: Err(Remote User Terminated Connection)
INFO - [gatt] task finished
INFO - [adv] advertising
```

But I had to change they heap allocation - otherwise the bootloader complained about .bss overlapping the bootloader RAM
(
```rust
    //esp_alloc::heap_allocator!(size: 72 * 1024);
    esp_alloc::heap_allocator!(#[link_section = ".dram2_uninit"] size: 64 * 1024);
```
)

(And the original BLE examples in esp-hal also work fine for me)

The backtrace also looks "interesting"

HaoboGu, 2025-04-14:
I had similar error when enabling "defmt". The app struck at "Disabling RNG early entropy source", no more responses.

I have to switch to use "log" feature, which works well for me.

artisdom, 2025-04-15:
> 
> But I had to change they heap allocation - otherwise the bootloader complained about .bss overlapping the bootloader RAM (
> 
>     //esp_alloc::heap_allocator!(size: 72 * 1024);
>     esp_alloc::heap_allocator!(#[link_section = ".dram2_uninit"] size: 64 * 1024);
> 
> (And the original BLE examples in esp-hal also work fine for me)
> 
> The backtrace also looks "interesting"

Pull the latest code today, and run again, my previously reported issue seems gone,
saw this same issue on ESP32 Wrover Kit now.

```
cargo +esp run --release --no-default-features --features=esp32 --target=xtensa-esp32-none-elf --bin ble_bas_peripheral

   Compiling esp-metadata v0.6.0 (https://github.com/esp-rs/esp-hal.git?rev=56be259c41305f24276852d2af4fce16247107bd#56be259c)
   Compiling esp-config v0.3.1 (https://github.com/esp-rs/esp-hal.git?rev=56be259c41305f24276852d2af4fce16247107bd#56be259c)
   Compiling esp-hal v1.0.0-beta.0 (https://github.com/esp-rs/esp-hal.git?rev=56be259c41305f24276852d2af4fce16247107bd#56be259c)
   Compiling esp-hal-embassy v0.7.0 (https://github.com/esp-rs/esp-hal.git?rev=56be259c41305f24276852d2af4fce16247107bd#56be259c)
   Compiling esp-wifi v0.13.0 (https://github.com/esp-rs/esp-hal.git?rev=56be259c41305f24276852d2af4fce16247107bd#56be259c)
   Compiling esp-backtrace v0.15.1 (https://github.com/esp-rs/esp-hal.git?rev=56be259c41305f24276852d2af4fce16247107bd#56be259c)
   Compiling toml v0.8.20

   Compiling embassy-embedded-hal v0.3.0
   Compiling bt-hci v0.3.0
   Compiling trouble-example-apps v0.1.0 (C:\r\trouble\examples\apps)
    Finished `release` profile [optimized + debuginfo] target(s) in 2m 24s
     Running `espflash flash --monitor target\xtensa-esp32-none-elf\release\ble_bas_peripheral`
[2025-04-15T02:35:02Z INFO ] Detected 5 serial ports
[2025-04-15T02:35:02Z INFO ] Ports which match a known common dev board are highlighted
[2025-04-15T02:35:02Z INFO ] Please select a port
[2025-04-15T02:40:52Z INFO ] Serial port: 'COM61'
[2025-04-15T02:40:52Z INFO ] Connecting...
[2025-04-15T02:40:52Z INFO ] Using flash stub
Chip type:         esp32 (revision v1.0)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, Coding Scheme None
MAC address:       ac:67:b2:71:63:68
App/part. size:    344,096/4,128,768 bytes, 8.33%
[2025-04-15T02:40:54Z INFO ] Segment at address '0x1000' has not changed, skipping write
[2025-04-15T02:40:54Z INFO ] Segment at address '0x8000' has not changed, skipping write
[00:00:21] [========================================]     205/205     0x10000                                                                                                                                                               [2025-04-15T02:41:16Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7104
load:0x40078000,len:15576
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3876
entry 0x4008064c
0x4008064c - core::fmt::Arguments::new_v1
    at C:\Users\Ted\.rustup\toolchains\esp\lib\rustlib\src\rust\library\core\src\fmt\mod.rs:604
I (30) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (30) boot: compile time Jun  7 2023 07:48:23
I (32) boot: Multicore bootloader
I (36) boot: chip revision: v1.0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
[... 116 lines cut ...]
I (127) esp_image: segment 3: paddr=0001b2cc vaddr=40080000 size=04d4ch ( 19788) load
I (143) esp_image: segment 4: paddr=00020020 vaddr=400d0020 size=3dfd8h (253912) map
I (235) esp_image: segment 5: paddr=0005e000 vaddr=40084d4c size=05ff0h ( 24560) load
I (251) boot: Loaded app from partition at offset 0x10000
I (251) boot: Disabling RNG early entropy source...
INFO - Our address = Address { kind: AddrKind(1), addr: BdAddr([255, 143, 26, 5, 228, 255]) }
INFO - Starting advertising and GATT service
INFO - [host] filter accept list size: 12
INFO - [host] setting txq to 10, fragmenting at 251
INFO - [host] configuring host buffers (8 packets of size 255)
INFO - [host] initialized
INFO - [host] Device Address AE:67:B2:71:63:69
INFO - [adv] advertising
INFO - [adv] connection established
INFO - [custom_task] notifying connection of tick 1
INFO - [custom_task] RSSI: 4
INFO - [custom_task] notifying connection of tick 2
INFO - [custom_task] RSSI: -51
INFO - [custom_task] RSSI: -41
INFO - [gatt] Read Event to Level Characteristic: Ok(60)
INFO - [custom_task] notifying connection of tick 61
INFO - [custom_task] RSSI: -44
```

Thanks guys. @bjoernQ @lulf

Dominaezzz, 2025-04-26:
I suppose this can be closed?

closed 2025-04-27 by artisdom