embassy-rs/trouble #346: Get CCCD with ATT_FIND_INFORMATION_REQ instead of ATT_READ_BY_TYPE_REQ

simonft, 2025-04-07:
[Caveat to all of the below: I'm newish to Rust, and very new to embedded development and Bluetooth]

I've been debugging why communication with a [third party peripheral](https://combustion.inc/products/predictive-thermometer-gen2-wifi-display) has been hanging on a call to characteristic_by_uuid() when other Bluetooth stack implementations (Bluez, iOS, Android) work. I narrowed down the issue to [get_characteristic_cccd()](https://github.com/embassy-rs/trouble/blob/851909523dc4d18dde192732c2f9ae795269559b/host/src/gatt.rs#L734) and have discovered the peripheral is not responding to ATT_READ_BY_TYPE_REQ when the attribute type is set to 0x2902 (the client characteristic configuration descriptor). It will respond when the attribute type is set to some others, like 0x2803 (Characteristic). If I'm reading the GATT spec correctly this is a trouble bug.

The spec [says](https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Core-54/out/en/host/generic-attribute-profile--gatt-.html#UUID-b5a253e2-98a4-63df-fc1a-e4555ec83ddc_table-idm13358912755578) "Only those portions of the ATT protocol requests, responses, notifications or indications necessary to implement the mandatory or supported optional sub-procedures is required." 

[4.6.2](https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Core-54/out/en/host/generic-attribute-profile--gatt-.html#UUID-b676406c-3e78-696a-29e8-e0f025f34426) describes using ATT_READ_BY_TYPE_REQ with the type set to Characteristic to retrieve the characteristics in a service, which is what trouble is doing. But for discovering all characteristics ([4.7.1](https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Core-54/out/en/host/generic-attribute-profile--gatt-.html#UUID-ca708893-b226-d29d-7308-bece78a8fe9c)) the spec uses ATT_FIND_INFORMATION_REQ, whereas trouble is using ATT_READ_BY_TYPE when it wants to find the Client Characteristic Configuration Descriptor. This makes sense in that trouble only cares about one type of descriptor at that point, but from my read the GATT spec doesn't provide a way to get just one type of descriptor.

My takeaway is that trouble should be changed to use ATT_FIND_INFORMATION_REQ to get the handle of the Client Characteristic Configuration, but that's based on my understanding that a server is only required to implement the sub-parts of the various PDUs that are actually used in the diagrams in the GATT spec, which might be incorrect.

FWIW, my read of the [relevant Bluez code](https://github.com/bluez/bluez/blob/master/src/shared/gatt-client.c#L791) is that it uses ATT_FIND_INFORMATION_REQ unless there's only one possible place the CCCD could be, in which case it just returns that handle.

If all of the above is correct I'm happy to take a crack at a PR for this, though as noted in the caveat at the top I'm relatively new to multiple pieces of this.

And I don't think it's relevant, but in case it's helpful I'm using the latest commit from the main branch on an ESP32c3.

HaoboGu, 2025-04-08:
It seems that the peripheral implementation has similar issue. When connecting Linux as peripheral, Linux only sends `ATT_FIND_INFORMATION_REQ` to get CCCD, so TrouBLE doesn't set CCCD value in this case. I suspect that my notification failing is related.

[btsnoop.log](https://github.com/user-attachments/files/19641241/tmp2.log)

lulf, 2025-04-10:
Thanks for taking the time to investigate and for the detailed bug report. Your reasoning and reading of the spec seems correct to me. I think it's worth making the change and we can see how it behaves for the tests.

simonft, 2025-04-10:
I opened a draft PR. I haven't figured out how to get the gatt_client_server tests running locally though (I guess I need to set Zephyr up somehow?) and I think a maintainer probably has to kick of the GitHub Actions ones.

I did test it with an actual peripheral and it worked for that.

simonft, 2025-05-20:
I was away from a peripheral I could test with for a month, but I'm taking a look at this again. The tests on the PR are failing, but it's not obvious to me if there's a way to run them locally, or if there's another way I can debug or get more information about the error. Is that documented somewhere?

PR #350 (2025-08-07): Fix: Get cccd with FIND_INFORMATION instead of READ_BY_TYPE
Fixes #346
closed 2025-08-07 by lulf