embassy-rs/trouble #371: Invalid PSM uses in L2CAP example

bittailor, 2025-04-24:
Hi 

I think the current LE L2CAP examples use an _invalid_ `PSM` value. 

https://github.com/embassy-rs/trouble/blob/2128a22360300554b90efaa7804c759b24a60fad/examples/apps/src/ble_l2cap_peripheral.rs#L54

It works when both, central and peripheral, are done with trouble on an embedded device. But when I try to connect from macOS as central (using CoreBluetooth in a Swift test client) to a peripheral done with the trouble example code, the CoreBluetooth call 

``` swift
peripheral.openL2CAPChannel(0x2349)
```

fails with

```
Failed to open L2CAP channel Optional(Error Domain=CBErrorDomain Code=1 "One or more parameters were invalid." UserInfo={NSLocalizedDescription=One or more parameters were invalid.})
```

When looking at the [Bluetooth Specification for the PSM (or rather SPSM as it's called for LE L2CAP)](https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Core-60/out/en/host/logical-link-control-and-adaptation-protocol-specification.html#UUID-1ffdf913-7b8a-c7ba-531e-2a9c6f6da8fb) it should be in the range of `0x0000-0x00FF` or rather `0x0080-0x00FF` for one not assigned by SIG.

[image]

So when I change the PSM from `0x2349` to `0x81` in the trouble example 
```
let mut ch1 = L2capChannel::accept(&stack, &conn, &[0x2349], &Default::default())
                .await
                .unwrap();
```

and also in my Swift test client, the CoreBluetooth call

``` swift
peripheral.openL2CAPChannel(0x81)
```
Succeeds and I can exchange data.

lulf, 2025-04-25:
Huh, I didn't know about that restriction. Can you open a PR to fix it?

bittailor, 2025-04-25:
> Huh, I didn't know about that restriction. Can you open a PR to fix it?

👍 will do so.

bittailor, 2025-04-29:
#377

closed 2025-04-29 by bittailor