embassy-rs/trouble #372: Panic after pairing request

Erik1000, 2025-04-25:
I am using the nrf Connect App on iOS. After connecting, requesting read on a characteristic which is rejected with `AttErrorCode::INSUFFICIENT_ENCRYPTION`, iOS automatically asks for pairing. After clicking "OK", the trouble-host panics:

```
====================== PANIC ======================
panicked at /home/erik/.cargo/git/checkouts/trouble-c44d851c1359feda/2128a22/host/src/host.rs:311:27:
range end index 65 out of range for slice of length 27

Backtrace:

0x4205c4ba
core::slice::index::slice_end_index_len_fail::do_panic::runtime
    at /rustc/05f9846f893b09a1be1fc8560e33fc3c815cfecb/library/core/src/panic.rs:218
0x4205b5f6
core::slice::index::slice_end_index_len_fail::do_panic
    at /rustc/05f9846f893b09a1be1fc8560e33fc3c815cfecb/library/core/src/intrinsics/mod.rs:3886
0x4203f392
<core::ops::range::Range<usize> as core::slice::index::SliceIndex<[T]>>::index_mut
    at /home/erik/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/slice/index.rs:449
0x4202657e
trouble_host::host::RxRunner<C,P>::run_with_handler::{{closure}}
    at /home/erik/.cargo/git/checkouts/trouble-c44d851c1359feda/2128a22/host/src/host.rs:661
0x4203a7d4
<embassy_futures::select::Select3<A,B,C> as core::future::future::Future>::poll
    at /home/erik/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/embassy-futures-0.1.1/src/select.rs:111
0x42030490
trouble_host::host::Runner<C,P>::run_with_handler::{{closure}}
    at /home/erik/.cargo/git/checkouts/trouble-c44d851c1359feda/2128a22/host/src/host.rs:612
0x42033990
car_starter::____embassy_main_task::{{closure}}
    at /home/erik/Documents/car/repo/starter-firmware/src/main.rs:55
0x42040574
embassy_executor::raw::SyncExecutor::poll::{{closure}}
    at /home/erik/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/embassy-executor-0.7.0/src/raw/mod.rs:430
0x420378b8
esp_hal_embassy::executor::thread::Executor::run
    at /home/erik/.cargo/git/checkouts/esp-hal-29ef8683203c9b81/8f0c517/esp-hal-embassy/src/executor/thread.rs:111
0x4203c58c
main
```

I am using an Esp32C3 as microcontroller with embassy.

HaoboGu, 2025-04-25:
Looks like the mtu config is too small. Try to enable `l2cap-default-packet-pool-mtu-255` feature?

lulf, 2025-04-25:
Should be fixed on the main branch now, verified the examples work with the new out of the box configuration. Reopen if you still have issues.

closed 2025-04-25 by lulf

lulf, 2025-04-25:
At the moment, yes. The reason is that users can choose to store this differently depending on the application.