embassy-rs/trouble #375: Allow as_gatt() to support owned or caller-provided serialization buffers

sssilver, 2025-04-28:
Currently, the AsGatt trait defines:

```rs
fn as_gatt(&self) -> &[u8];
```

This intends for implementors to return a reference to serialized data. However, it creates challenges:
- It requires the serialized form to be pre-stored inside the struct
- It prevents on-demand serialization without unsafe code
- It makes it harder to implement serialization for dynamic or large payloads efficiently

Because the function must return a reference to a byte array, we can't safely serialize into a temporary buffer inside `as_gatt()` without storing that buffer somewhere in the struct itself, increasing memory usage even when serialization is infrequent.

This also complicates patterns like serialization depending on runtime field values, serialization into external/caller-provided memory to save RAM, or avoiding stale serialized buffers when fields change.

Would it be possible to provide an alternative trait or extend AsGatt with a variant that returns an owned serialization result? E.g.:

```rs
fn to_gatt(&self) -> heapless::Vec<u8, N>;
```
or something like:
```rs
fn to_gatt(&self, buffer: &mut [u8]) -> usize;
```

This would allow on-demand serialization without extra storage inside the object itself.

Dominaezzz, 2025-05-20:
Perhaps `fn as_gatt(&self) -> &[u8];` could be `fn as_gatt(&self) -> impl AsRef<[u8]>;`

sssilver, 2025-05-21:
I may be misunderstanding something, but I don't see how that solves the issue. 

The underlying problem is lifetimes and memory ownership—not just the trait signature. The issue is that as_gatt(&self) must return a reference (&[u8]), which requires the bytes to live somewhere tied to self, e.g., precomputed and stored inside the struct. Changing the return type to impl AsRef<[u8]> doesn’t change that.

Developers would still not be able to return a temporary buffer created inside the function without allocating or storing it somewhere.

Dominaezzz, 2025-05-21:
> Developers would still not be able to return a temporary buffer created inside the function

Yes you would. Arrays implement AsRef.
You'd be able to return a temporary array

sssilver, 2025-05-21:
I stand corrected then -- sounds great!

Dominaezzz, 2025-05-24:
I want to make a PR for this but I'm not sure what's best to return between `AsRef<[u8]>`, `Borrow<[u8]>` or `Deref<Target=[u8]>`.

Dominaezzz, 2025-05-25:
Sadly https://github.com/embassy-rs/trouble/blob/925d66d94654451257e9741ee76341e5c749b091/host/src/attribute.rs#L603 doesn't like temporaries. Looks like a non-trivial refactor is needed.