embassy-rs/trouble #379: How to know when a notify characteristic has been subscribed to?

Dominaezzz, 2025-04-29:
I see notifying central works [like so](https://github.com/embassy-rs/trouble/blob/7d72e8d0cf4558bb54f0a1ab43cdb88fabe160a8/examples/apps/src/ble_bas_peripheral.rs#L180-L203) but is there a way to suspend that loop when there is no subscription?
In my application, the loop is rather expensive and I only want to do it if the client is asking for it.

My question is, is this feature currently available and if not where would it make sense to expose it, like a `GattEvent` perhaps?

HaoboGu, 2025-04-29:
Check the cccd value of this characteristic

Dominaezzz, 2025-04-29:
Right, thanks for the pointer!

I found this [CCCD](https://github.com/embassy-rs/trouble/blob/7d72e8d0cf4558bb54f0a1ab43cdb88fabe160a8/host/src/attribute.rs#L876) struct with a `should_notify` function on it.

I also found this [`cccd_handle()`](https://github.com/embassy-rs/trouble/blob/7d72e8d0cf4558bb54f0a1ab43cdb88fabe160a8/host/src/attribute.rs#L681C12-L681C23) on the `Characteristic` struct.

I'm trying to figure out how to get from one to the other.

It also seems like the CCCD value only tells me if a characteristic supports notifications, it does seem to say whether the client has made a subscription?

Dominaezzz, 2025-05-03:
I'm stumped, I wasn't able to figure how to get the CCCD value using the handle 😅

lulf, 2025-05-03:
> I see notifying central works [like so](https://github.com/embassy-rs/trouble/blob/7d72e8d0cf4558bb54f0a1ab43cdb88fabe160a8/examples/apps/src/ble_bas_peripheral.rs#L180-L203) but is there a way to [...]

I think this makes sense, some kind of GattEvent::Subscribed { handle } and GattEvent::Unsubscribed { handle } event. 

I'm not sure there is an API for getting the CCCD value, but I think the above should be a nice abstraction for what you need.

HaoboGu, 2025-05-03:
the `cccd_handle` of a characteristic is public:

https://github.com/embassy-rs/trouble/blob/c675b4d5f56350cc899d8f847324aad7db29c27a/host/src/attribute.rs#L618

And `GattConnectionEvent::Gatt { event: GattEvent::Write}` can be monitored for cccd write event.

Dominaezzz, 2025-05-03:
Ohh I see, so rather than try to read the current value of the CCCD, I can just monitor the writes to it, and get the value from the event.
Okay that does unblock me at least.

It would still be nice to have a way to direct just read the current CCCD at least.

HaoboGu, 2025-05-03:
> It would still be nice to have a way to direct just read the current CCCD at least.

Yes you can, use `characteristic.cccd_handle` for reading a single CCCD value for a characteristic and use `att_server.get_cccd_table` for getting to full CCCD table

Dominaezzz, 2025-05-03:
> Yes you can, use `characteristic.cccd_handle` for reading a single CCCD value

How? Just to make sure I'm not confused, is the handle itself actually the value? Or do I need to call something else to read the value being the handle?

Dominaezzz, 2025-05-03:
Oohh okay I understand now. So I can do `server.server.get_cccd_table(&connection).inner()` and search the tuple pairs with the `cccd_handle`.

Thanks!

I'll close this issue for now as my issue is solved.

closed 2025-05-03 by Dominaezzz