embassy-rs/trouble #405: Can't receive periodic advertising

Tomer-Eliahu, 2025-06-02:
I am using the Micro:bit as a receiver for a broadcast. 

I can correctly get the extended advertising info using extended scan, but I can't seem to then be able to receive periodic advertising from that device. I am getting an adv_interval value of 144 in the extended advertising, so there definitely is periodic advertising from that device as well.

I have made a fork of trouble trying to add support for using the HCI command [7.8.67. LE Periodic Advertising Create Sync command](https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Core-54/out/en/host-controller-interface/host-controller-interface-functional-specification.html). The full changes are [here](https://github.com/embassy-rs/trouble/compare/main...Tomer-Eliahu:trouble:patch-1).

Essentially, I have added

```rust
                            LeEvent::LePeriodicAdvertisingReport(data) => {
                                #[cfg(feature = "scan")]
                                {
                                    event_handler.on_periodic_adv_report(data);
                                    //The data is a *single* report as specified here 
                                    //https://docs.rs/bt-hci/latest/bt_hci/event/le/struct.LePeriodicAdvertisingReport.html
                                }
                            }
```

to the `run_with_handler` function and the following function to the scanner:

```rust 
pub async fn scan_periodic(&mut self, config: &ScanConfig<'_>, param: LePeriodicAdvCreateSyncParams) -> Result<ScanSession<'_, true>, BleHostError<C::Error>>
    where
        C: ControllerCmdAsync<LePeriodicAdvCreateSync> //Note the Async variation here
            + ControllerCmdSync<LeClearPeriodicAdvList>
            + ControllerCmdSync<LeAddDeviceToPeriodicAdvList>
            + ControllerCmdSync<LeSetExtScanEnable> //We also enable extended scanning
            + ControllerCmdSync<LeSetExtScanParams>
            + ControllerCmdSync<LeClearFilterAcceptList>
            + ControllerCmdSync<LeAddDeviceToFilterAcceptList>,
    {
        //First we enable extended scanning.
        let host = &self.central.stack.host;
        let drop = crate::host::OnDrop::new(|| {
            host.scan_command_state.cancel(false);
        });
        host.scan_command_state.request().await;
        self.central.set_accept_filter(config.filter_accept_list).await?;

        let scanning = ScanningPhy {
            active_scan: config.active,
            scan_interval: config.interval.into(),
            scan_window: config.window.into(),
        };

        let phy_params = crate::central::create_phy_params(scanning, config.phys);
        let host = &self.central.stack.host;
        host.command(LeSetExtScanParams::new(
            host.address.map(|s| s.kind).unwrap_or(AddrKind::PUBLIC),
            if config.filter_accept_list.is_empty() {
                bt_hci::param::ScanningFilterPolicy::BasicUnfiltered
            } else {
                bt_hci::param::ScanningFilterPolicy::BasicFiltered
            },
            phy_params,
        ))
        .await?;

        host.command(LeSetExtScanEnable::new(
            true,
            FilterDuplicates::Disabled,
            config.timeout.into(),
            bt_hci::param::Duration::from_secs(0),
        ))
        .await?;

        //This is the difference from simply scan_ext
        host.async_command(LePeriodicAdvCreateSync::new(
            param.options,
            param.adv_sid,
            param.adv_addr_kind,
            param.adv_addr,
            param.skip,
            param.sync_timeout,
            param.sync_cte_kind
        ))
        .await?;

        drop.defuse();
        Ok(ScanSession {
            command_state: &self.central.stack.host.scan_command_state,
            deadline: if config.timeout.as_ticks() == 0 {
                None
            } else {
                Some(Instant::now() + config.timeout.into())
            },
            done: false,
        })
    }
```

I am not getting an HCI error after calling this function and my code simply waits for an event that never seems to come.

Am I missing something else I need to add to get this functionality?

I am using the `nrf-sdc` as the controller if that matters (but only through trouble).

Tomer-Eliahu, 2025-06-03:
I am new to BLE so I might be missing something obvious.

I have the following config when building the sdc: 

```rust

        .support_ext_central()? //includes supporting extended scan
        .support_ext_adv()?
        .support_dle_central()?
        .support_le_2m_phy()? 
        .support_le_coded_phy()? 
        .support_phy_update_central()? 
        .support_le_periodic_adv()?
        .support_le_periodic_sync()?
        .support_periodic_adv_sync_transfer_receiver_central()? //I think this is not needed but just in case
        .support_sca_central()? 
        .central_count(1)?
        //It is vital we add the following counts as all but adv_count are default 0
        // I also tried setting all the counts to 2.
        .adv_count(1)?
        .periodic_adv_count(1)?
        .periodic_sync_count(1)?
```

I have changed the following code: 

```rust
        LeSetEventMask::new(
            LeEventMask::new()
                .enable_le_conn_complete(true)
                .enable_le_enhanced_conn_complete(true)
                .enable_le_conn_update_complete(true)
                .enable_le_adv_set_terminated(true)
                .enable_le_adv_report(true)
                .enable_le_scan_timeout(true)
                .enable_le_ext_adv_report(true)
                .enable_le_periodic_adv_sync_established(true) //added
                .enable_le_periodic_adv_report(true) //added
                .enable_le_periodic_adv_sync_lost(true)//added
                //.enable_le_periodic_adv_sync_established_v2(true)// --also attempted
                //.enable_le_periodic_adv_report_v2(true) //--also attempted
                .enable_le_long_term_key_request(true)
                .enable_le_phy_update_complete(true),
        )
```

It still doesn't work. Any ideas on what else to try? Thank you very much for your help!

Tomer-Eliahu, 2025-06-05:
I added support for printing the sync established event and surprisingly, the controller does actually sync to the PA but never
generates a report. I get this:

```
 The data is LePeriodicAdvertisingSyncEstablished {
     status: Ok(
         (),
    ),
    sync_handle: SyncHandle(
         0,
    ),
     adv_sid: 1,
    adv_addr_kind: AddrKind(
       1,
    ),
    adv_addr: BdAddr(
        [146, 30, 164, 142, 131,38,],
   ),
   adv_phy: Le2M,
    periodic_adv_interval: Duration(
        144,
     ),
    adv_clock_accuracy: Ppm50,
 }
```

davidmweber, 2025-06-21:
The scanner really needs some sort of fire and forget like your scan_periodic trait. I managed to get a variant of the code in the example ble_scanner app but had to put the scan() trait into a loop and call it repeatedly to get up to date scans. Have you (@Tomer-Eliahu ) made a PR for your periodic scanner? 

There is also a bug in the current scanner in the trouble-host/src/scan.rs file at line 114. I don't think this material but it needs to be fixed.

```rust
        let params = LeSetScanParams::new(
            if config.active {
                bt_hci::param::LeScanKind::Active
            } else {
                bt_hci::param::LeScanKind::Passive
            },
            config.interval.into(),  
            config.interval.into(),   // <--- Should read "config.window.into()"
            host.address.map(|a| a.kind).unwrap_or(AddrKind::PUBLIC),
            if config.filter_accept_list.is_empty() {
                bt_hci::param::ScanningFilterPolicy::BasicUnfiltered
            } else {
                bt_hci::param::ScanningFilterPolicy::BasicFiltered
            },
        );
 
```

davidmweber, 2025-06-22:
What hardware do you need to test on?  I have ESP32-C6 and ESP32-S3 at hand and can do tests if that will help you.

Tomer-Eliahu, 2025-08-12:
Slight update: 
 
I did a sanity check and using the nrf-connect sdk (version 3.0.2), I can correctly sync (and get periodic adv report info) with the same broadcasting device and board.

When I looked just now at the [softdevice controller changelog](https://github.com/nrfconnect/sdk-nrfxlib/blob/main/softdevice_controller/CHANGELOG.rst#bug-fixes), I saw this:

> Fixed an issue where the controller would stop generating advertising reports while running an extended scanner. (DRGN-25859).

Note this was under version 3.1.0, but since no version 3.0.2 is mentioned maybe this was incorporated into the 3.0.2 sdk?

When I previously looked (about 3 weeks ago), there was no mention of bugs that could explain this issue.

So, this might be a result of using the softdevice controller bindings which use the [version 2.7 branch of nrfx-lib](https://github.com/alexmoon/nrf-sdc/blob/main/.gitmodules) or it is a problem somewhere in trouble itself.

I did see a passing dev comment on the Apache-nimble code base that if some buffer (I believe it was an ACL buffer, but I am not sure) was too big, it might result in the periodic advertising report event never being generated (I can't seem to find that comment now though).