embassy-rs/trouble #408: running examples using esp-generate

jrjdavidson, 2025-06-12:
I wanted to try esp-generate V4 with the new trouble. All worked well until i tried to implement one of the examples. The compiler complained with confusing messages. After a bit of headscratching, i was able to run the example by:
- pointing my cargo.toml 'trouble-host' to the latest git, as the crate is out of date
- updating the bt-hci crate to the latest version (0.3.2)

Hope this helps.

jamessizeland, 2025-06-13:
@lulf are we able to publish a 0.2 soon?

lulf, 2025-06-13:
Yeah, i guess its time!

jamessizeland, 2025-06-13:
I'm not sure, but you would still always have the option of pulling trouble from GitHub still anyway, so that should be okay?

jamessizeland, 2025-06-13:
Our 0.1 is currently behind on released embassy anyway, hence the desire for another release.

chrissuozzo, 2025-06-15:
Yeah, it's all good; I'm working on a tutorial for `trouble-host`/BLE that's coming out in a few weeks which had been based on v0.1, but it would be nice to show off the latest & greatest if that will be out soon!

jamessizeland, 2025-06-15:
Then we'll definitely get it polished and published for you! Please feel free to use any bits from these repos I've made for introducing trouble:
 https://github.com/jamessizeland/microbit-ble-gamepad

https://github.com/jamessizeland/esp32c3-devkit-demo

And my favourite helpful feature is the UUID constants in here: https://github.com/embassy-rs/bt-hci/blob/main/src%2Fuuid%2Fcharacteristic.rs

closed 2025-07-15 by lulf

jrjdavidson, 2025-10-05:
looks like this has been updated, see 
https://github.com/esp-rs/esp-generate/pull/231