embassy-rs/trouble #423: Update `rand_core` to 0.9

HaoboGu, 2025-07-15:
I found that both `trouble-host` and examples use `rand_core = 0.6`. Is there any particular reason to not bump version to latest?

AndWass, 2025-07-15:
I made a quick attempt but one of the dependencies of p256 is stuck on rand_core 0.6.4 so you need that version to get the correct trait versions.

lure23, 2025-07-18:
Their [version 0.14 (unreleased) CHANGELOG](https://github.com/RustCrypto/elliptic-curves/blob/86e55d0a94fe7aa531356823e393a6d6f08deda1/sm2/CHANGELOG.md?plain=1#L14) lists: 

>Update to `rand_core` v0.9 ([diff](https://github.com/RustCrypto/elliptic-curves/pull/1125/files))

So the issue becomes pushing for a crate release, [New crate versions?](https://github.com/RustCrypto/elliptic-curves/issues/1068)