embassy-rs/trouble #429: Issue with multiple characteristics on the same UUID.

EmilNorden, 2025-07-25:
Hello,

I am relatively new in embedded Rust and Bluetooth development as well, so forgive me if I am making incorrect assumptions.

I am trying to connect to a gamepad ([8BitDo Ultimate 2C Wireless](https://www.8bitdo.com/ultimate-2c-wireless-controller/)) and have been basing my attempts on the ble_bas_central example. My problem is that this gamepad has multiple report characteristics for UUID 0x2A4D, and if I am not mistaken [GattClient::characteristic_by_uuid]( https://github.com/embassy-rs/trouble/blob/f715697214f364792953f9263d0ba6e28d2b36fd/host/src/gatt.rs#L759) will return the first characteristic it finds for the given UUID. Turns out that in order to read gamepad button presses, I need to subscribe to the **second** characteristic out of the 3 that exists for the UUID.

Is this by design or should there not be a way to retrieve multiple characteristics for a UUID? Or is there another way I can more accurately pinpoint the relevant characteristic with trouble?

Thanks.

EmilNorden, 2025-07-28:
Hi! Thank you for your reply!

Having multiple characteristics with the same UUID seems to be according to the specification:
[Section 3.3.1 in the GATT specification](https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Core-54/out/en/host/generic-attribute-profile--gatt-.html#UUID-cc38bdec-c550-d1a5-dacd-b9c44345d08d)

> A service may have multiple characteristic definitions with the same Characteristic UUID.

About needing the descriptor - this part I am not sure of and will need help to understand from you guys who are more familiar with bluetooth in general. According to [4.7.1 Discover All Characteristic Descriptors](https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Core-54/out/en/host/generic-attribute-profile--gatt-.html#UUID-ca708893-b226-d29d-7308-bece78a8fe9c):

> The ATT_FIND_INFORMATION_REQ PDU shall be used with the Starting Handle set to the handle of the specified characteristic value + 1 and the Ending Handle set to the ending handle of the specified characteristic.

The starting handle is easy. That one is given by the characteristic in question. But I am not sure of the ending handle. Is there such a thing as an explicit "ending handle"? I assumed I needed to find the declaration handle of the *next* characteristic declaration and subtract 1, since they are laid out sequentially.

My problem right now is, say I submit a PR that extend the GATT API to provide a way to get *all* characteristics of a given UUID. I would still need a way to get the descriptors (using the "find information" request) for each of these characteristics. How would such an API look if my assumptions about "ending handle" above is correct?

alexmoon, 2025-07-28:
Yes, Characteristic UUIDs identify a *kind* of characteristic, not an *instance* of a characteristic. Depending on the kind of service, multiple characteristics with the same UUID can be common. This is particular the case for the HID service, where you have a characteristic with the same UUID for each HID report.

I'd probably recommend changing the function to be `characteristics_by_uuid() -> Vec<...>` to match the `services_by_uuid`, then maybe add a convenience function like `first_characteristic_by_uuid()`.