embassy-rs/trouble #433: Unexpected GATT Event Behavior with Long Characteristic Values on ESP32-S3

LeonRein, 2025-07-31:
I am experimenting with the [ble\_bas\_peripheral.rs](https://github.com/embassy-rs/trouble/blob/main/examples/esp32/src/bin/[ble_bas_peripheral.rs](https://github.com/embassy-rs/trouble/blob/main/examples/esp32/src/bin/ble_bas_peripheral.rs)) example on an ESP32-S3.

I have added a `HeaplessString<255>` as a new characteristic to the `BatteryService`:

```rust
#[gatt_service(uuid = service::BATTERY)]
struct BatteryService {
    /// Battery Level
    #[descriptor(uuid = descriptors::VALID_RANGE, read, value = [0, 100])]
    #[descriptor(uuid = descriptors::MEASUREMENT_DESCRIPTION, name = "hello", read, value = "Battery Level")]
    #[characteristic(uuid = characteristic::BATTERY_LEVEL, read, notify, value = 10)]
    level: u8,
    #[characteristic(uuid = "408813df-5dd4-1f87-ec11-cdb001100000", write, read, notify)]
    status: bool,
    #[characteristic(uuid = "1234", read, write, notify)]
    custom: String<255>,
}
```

I also modified the default `GattEvent` handling to print the string value:

```rust
GattEvent::Read(event) => {
    if event.handle() == custom.handle {
        let value = server.get(&custom).unwrap_or(String::try_from("Could not parse value to String").unwrap());                            
        info!("[gatt] Read Event to Level Characteristic: {}", value.as_str());
    }
}
GattEvent::Write(event) => {
    if event.handle() == custom.handle {
        let value = event.value(&custom).unwrap_or(String::try_from("Could not parse value to String").unwrap());     
        info!("[gatt] Write Event to Level Characteristic: {}", value.as_str());
    }
}
```

I can read and write this characteristic with the nRF Connect app.
When I use a 10-byte message, everything works as expected, and both events appear in the log:

```
INFO - [gatt] Write Event to Level Characteristic: 1234567890
INFO - [custom_task] notifying connection of tick 4
INFO - [custom_task] RSSI: -44
INFO - [custom_task] notifying connection of tick 5
INFO - [custom_task] RSSI: -44
INFO - [gatt] Read Event to Level Characteristic: 1234567890
INFO - [custom_task] notifying connection of tick 6
INFO - [custom_task] RSSI: -41
```

However, when I send a message longer than the default MTU (e.g., a 30-byte message with an MTU of 23), the behavior gets strange:

* The Write Event does not appear in the log at all.
* I can still read the correct value back.
* The Read Event is triggered **twice**.

Example log:

```
INFO - [custom_task] notifying connection of tick 9
INFO - [custom_task] RSSI: -43
##### Expected Write Event here for value 123456789012345678901234567890 #####
INFO - [custom_task] notifying connection of tick 10
INFO - [custom_task] RSSI: -44
INFO - [custom_task] notifying connection of tick 11
INFO - [custom_task] RSSI: -43
INFO - [gatt] Read Event for Custom Characteristic: 123456789012345678901234567890
INFO - [gatt] Read Event for Custom Characteristic: 123456789012345678901234567890
INFO - [custom_task] notifying connection of tick 12
INFO - [custom_task] RSSI: -48
```

If I manually request a higher MTU with the nRF Connect app, the behavior becomes normal again. One Write Event and one Read Event.

My questions:

1. Is this behavior expected, or is it a bug?
2. Is there another way to reliably catch the Write Event for long messages?
3. Can the default MTU in trouble be changed?

I am asking because I want to use the Web Bluetooth API to communicate from a small website with the ESP32-S3. The Web Bluetooth API does not support custom MTU requests, but I need to transmit messages longer than 20–23 bytes.

alexmoon, 2025-07-31:
Characteristic values longer than the MTU require different procedures for read/write (specifically Vol 3 Part G 4.8.3 "Read Long Characteristic Values" and Vol 3 Part G 4.9.4 "Write Long Characteristic Values). I don't know if trouble supports those procedures properly yet. However, you will expect multiple PDUs for each read/write in those cases. The PDUs use offsets to identify which part of the characteristic is being read/written.

lulf, 2025-08-20:
> Thanks for pointing me in the right direction!
> 
> Writing a long characteristic value is split into multiple `PrepareWrite` events, followed by a final `ExecuteWrite` event. Reading a long characteristic value is split into a `Read` event and multiple `ReadBlob` events.
> 
> let payload = event.payload();
> let incoming = payload.incoming();
> info!("[gatt] AttClient: {:?}", incoming);
> 
> ```
> INFO - [gatt] Type: GattEvent::Other(_)
> INFO - [gatt] AttClient: Request(PrepareWrite { handle: 34, offset: 0, value: [49, 50, 51, 52, 53, 54, 55, 56, 57, 48, 49, 50, 51, 52, 53, 54, 55, 56] })
> INFO - [gatt] Type: GattEvent::Other(_)
> INFO - [gatt] AttClient: Request(PrepareWrite { handle: 34, offset: 18, value: [57, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 48] })
> INFO - [gatt] Type: GattEvent::Other(_)
> INFO - [gatt] AttClient: Request(ExecuteWrite { flags: 1 })
> INFO - [gatt] Type: GattEvent::Read(_)
> INFO - [gatt] AttClient: Request(Read { handle: 34 })
> INFO - [gatt] Type: GattEvent::Read(_)
> INFO - [gatt] AttClient: Request(ReadBlob { handle: 34, offset: 22 })
> ```
> 
> I find it a bit odd that both `Read` and `ReadBlob` events are classified as `GattEvent::Read(_)`, while `PrepareWrite` and `ExecuteWrite` events are both classified as `GattEvent::Other(_)`.
> 
> From what I understand, to retrieve the value of such a long write, I need to:
> 
>     1. Match the `AttClient` request as `AttClient::Request(AttReq::ExecuteWrite { .. })`.
> 
>     2. Accept the event so that the server processes the message.
> 
>     3. Then read the value back via `server.get(&value)`.
> 
> 
> Is this the correct approach? I wish there were an additional abstraction layer that would simply fire an event whenever a new value is fully received. Regardless of whether it came from a single write or multiple writes of a long value.

Agreed, that would be nice. I'm not sure what that abstraction would look like, but if you have something in mind, it would be great if you could propose it.